RT-Thread/rt-thread #2711: 为什么大家不到gitee.com上去讨论问题？

newflydd, 2019-05-22:
RT-Thread在gitee.com上也托管了项目代码。
但是gitee.com上几乎一个issue都没有，为什么大家不到中国的git上讨论问题？
美国已经技术封锁华为了，程序员也要有自己的备胎计划。
防止哪一天github被技术封锁了，要知道现在github可是美国企业微软的，微软是要服从美国政府管制的。

SummerGGift, 2019-05-22:
我猜分散起来不好管理吧

BernardXiong, 2019-05-22:
有道理。

另外，还有www.rt-thread.org国内论坛，那边也可以多多讨论。

MoonwalkerLee, 2019-05-23:
> RT-Thread在gitee.com上也托管了项目代码。
> 但是gitee.com上几乎一个issue都没有，为什么大家不到中国的git上讨论问题？
> 美国已经技术封锁华为了，程序员也要有自己的备胎计划。
> 防止哪一天github被技术封锁了，要知道现在github可是美国企业微软的，微软是要服从美国政府管制的。

按说应该是这样，但不知为什么我一登上gitee就感觉厌烦，整个页面乱糟糟的。。。

PengQianheng, 2019-05-23:
https://help.github.com/en/articles/github-and-export-controls

关于出口管制，GITHUB官网上有具体说明
可以参考一下

BernardXiong, 2019-05-23:
放心，RT-Thread和github出口管制没半毛钱关系

newflydd, 2019-05-23:
> 放心，RT-Thread和github出口管制没半毛钱关系

gitee.com又快又好，而且RT-Thread在gitee上一个issue都没有的情况下，gitee将RT-Thread评为最具价值的开源项目之一。
我还是喜欢国产的东西。
可以理解RT-Thread是冲着国际化和国际市场来的。
但并不是是说gitee就不能做到国际化，gitee的英文版也做到了100%翻译。

希望祖国以华为为代表的一片企业，一片产业都能走出国门，在国际市场与美帝分庭抗礼。

BernardXiong, 2019-05-23:
> > 放心，RT-Thread和github出口管制没半毛钱关系
> 
> gitee.com又快又好，而且RT-Thread在gitee上一个issue都没有的情况下，gitee将RT-Thread评为最具价值的开源项目之一。
> 我还是喜欢国产的东西。
> 可以理解RT-Thread是冲着国际化和国际市场来的。
> 但并不是是说gitee就不能做到国际化，gitee的英文版也做到了100%翻译。
> 
> 希望祖国以华为为代表的一片企业，一片产业都能走出国门，在国际市场与美帝分庭抗礼。

好的，我们后续会加强gitee那边的管理，并（尝试）进行双方的代码合并

Simon-Leung, 2019-05-24:
国内大都养肥了宰，以前中过coding的招，现在打死也不用国产的

BernardXiong, 2019-05-24:
coding.net也算能够理解，毕竟developer市场还是一块小市场。当有一定用户但没有好的商业模式时，就比较难熬了。coding.net最终被腾讯并购也算一种不错方式。

heyuanjie87, 2019-05-25:
趁github没被管制前多用用，不能轻易退缩，毕竟用坏了再换成备胎才是常规操作

BernardXiong, 2019-05-26:
@heyuanjie87 犀利 👍

closed 2019-06-28 by BernardXiong