RT-Thread/rt-thread #2714: stm32 hal can库问题

Simon-Leung, 2019-05-23:
CanTxMsgTypeDef在"rt-thread/bsp/stm32"里的hal库里，已经不支持。所用的新库的CAN_HandleTypeDef已经没有了成员CanTxMsgTypeDef和CanRxMsgTypeDef
新库的
`typedef struct __CAN_HandleTypeDef
{
  CAN_TypeDef                 *Instance;                 /*!< Register base address */

  CAN_InitTypeDef             Init;                      /*!< CAN required parameters */

  __IO HAL_CAN_StateTypeDef   State;                     /*!< CAN communication state */

  __IO uint32_t               ErrorCode;                 /*!< CAN Error code.
                                                              This parameter can be a value of @ref CAN_Error_Code */

} CAN_HandleTypeDef;`
旧库的
`typedef struct
{
  CAN_TypeDef                 *Instance;  /*!< Register base address          */

  CAN_InitTypeDef             Init;       /*!< CAN required parameters        */

  CanTxMsgTypeDef*            pTxMsg;     /*!< Pointer to transmit structure  */

  CanRxMsgTypeDef*            pRxMsg;     /*!< Pointer to reception structure */

  __IO HAL_CAN_StateTypeDef   State;      /*!< CAN communication state        */

  HAL_LockTypeDef             Lock;       /*!< CAN locking object             */

  __IO uint32_t               ErrorCode;  /*!< CAN Error code                 */

}CAN_HandleTypeDef;`

BernardXiong, 2019-05-23:
欢迎发一个PR补丁出来

zhangjun1996, 2019-06-28:
已经修复过了，这个议题也可以关了

closed 2019-06-28 by BernardXiong