RT-Thread/rt-thread #2724: rndis device收到REMOTE_NDIS_RESET_MSG消息后，网卡被禁用，无法再启动使用

cris8259, 2019-05-27:
如题。在保持网卡不断有通信请求的情况下，不断热插拔USB， usb device收到REMOTE_NDIS_RESET_MSG后，网卡无法查看属性信息，被禁用且无法再启动。可重现

BernardXiong, 2019-05-27:
请给出更多详细信息，RT-Thread版本，在哪个BSP上，工具链情况等

cris8259, 2019-05-27:
RTT3.1.2   stm32F407VE   lwip2.0.2  keilV5.18

BernardXiong, 2019-05-28:
@lymzzyh 大法师，看看这个呢，stm32的

cris8259, 2019-05-29:
[RNDIS] plugged out
[RNDIS] OID_GEN_CURRENT_PACKET_FILTER
[RNDIS] set
[RNDIS] OID_GEN_CURRENT_PACKET_FILTER
[RNDIS] set
[RNDIS] set
[RNDIS] OID_GEN_CURRENT_PACKET_FILTER
[RNDIS] set
[RNDIS] reset
//此时网卡无法通信，接着重新插拔一次
[RNDIS] plugged out
[RNDIS] OID_GEN_CURRENT_PACKET_FILTER
[RNDIS] set
[RNDIS] OID_GEN_CURRENT_PACKET_FILTER
[RNDIS] set
[RNDIS] set
[RNDIS] OID_GEN_CURRENT_PACKET_FILTER
[RNDIS] set
[RNDIS] reset

保持网卡有通信请求，插拔USB10次左右，会有reset消息，之后就无法通信，网卡属性无法查看，之后会被禁用。这个时候重新插拔usb也无法恢复，照样会收到reset消息。在设备管理器中usb设备正常，只是网卡不正常

cris8259, 2019-05-29:
@lymzzyh

cris8259, 2019-06-09:
发现没有对REMOTE_NDIS_RESET_MSG进行处理，找葫芦画瓢写了个REMOTE_NDIS_RESET_MSG的response函数。
static rt_err_t _rndis_reset_response(ufunction_t func,rndis_reset_msg_t msg)
{
	rndis_reset_cmplt_msg_t resp;
	struct rt_rndis_response * response;

	response = rt_malloc(sizeof(struct rt_rndis_response));
	resp = rt_malloc(sizeof(struct rndis_reset_cmplt_msg));

	if( (response == RT_NULL) || (resp == RT_NULL) )
	{
		RNDIS_PRINTF("%d: no memory!\r\n", __LINE__);

		if(response != RT_NULL)
			rt_free(response);

		if(resp != RT_NULL)
			rt_free(resp);

		return -RT_ENOMEM;
	}

	resp->MessageType = REMOTE_NDIS_RESET_CMPLT;
	resp->MessageLength = sizeof(struct rndis_reset_cmplt_msg);
	resp->Status = RNDIS_STATUS_SUCCESS;
	resp->AddressingReset = 0;
// 	switch (msg->Oid)
// 	{
// 	default:
// 		resp->Status = RNDIS_STATUS_FAILURE;
// 		return RT_EOK;
// 	}

	resp->Status = RNDIS_STATUS_SUCCESS;

	response->buffer = resp;

	{
		rt_base_t level = rt_hw_interrupt_disable();
		rt_list_insert_before(&((rt_rndis_eth_t)func->user_data)->response_list, &response->list);
		rt_hw_interrupt_enable(level);
	}

	return RT_EOK;
}

//对应处理的case
    case REMOTE_NDIS_RESET_MSG:
	ret = _rndis_reset_response(func,(rndis_reset_msg_t) msg);
        RNDIS_PRINTF("reset\n");
        break;
增加以上代码后，PC端网卡被禁用的问题似乎已解决，但是还是不能通信。有通信请求的情况下进行热插拔会出现有多个处于active 和 time-wait状态的tcp连接，而且一直存在。看代码发现每500ms会调用tcp_slowtmr()来清掉多种状态的tcp连接，下断点发现tcpip_tcp_timer(void *arg)根本不会周期性的执行，timer没有启动，不知道是什么原因

lymzzyh, 2019-06-10:
问题已修复 相关PR #2764 @cris8259 @BernardXiong

cris8259, 2019-06-10:
测试了一下，还是无法通信。现在都不需要热插拔，首次启动就无法通信了。首次启动会执行_function_enable，如果把_function_enable中的reset eth rx 和 tx部分去掉，则首次启动后可以通信，reset eth rx和tx 似乎有问题。@lymzzyh

cris8259, 2019-06-10:
刚开始是在V3.1.3版本，替换rndis.c文件，编译测试，有问题。后换成最新的4.0.2版本，也是一样有问题。一次热插拔就不行了，无法通信，有时候能ping通，每ping一下tcp数据通信能成功一次，似乎ping能激活一样。@lymzzyh

lymzzyh, 2019-06-11:
我这边目前是这样的情况 iperf 进行TCP链接带宽测试，当热插拔后需要收到RESET信息过后一小段时间后iperf链接能够自动恢复，并且ping工作正常。是否是应用层程序存在断连后不能正常通信的问题，而非网卡层有问题呢？

lymzzyh, 2019-06-11:
可以先通过iperf工具验证在你的平台上网卡这一层是否有问题。 @cris8259

BernardXiong, 2019-06-11:
@cris8259 因为这边验证的时候更多是使用github master版本来验证，所以可以先用最新版本一起来试试。

BernardXiong, 2019-06-14:
@cris8259 可以尝试下新的版本，如果依然有问题可以重开这个issue。

closed 2019-06-14 by BernardXiong