RT-Thread/rt-thread #2747: stm32f429如何实现SD卡热插拔

liucp2016, 2019-06-03:
如题， 求SD卡热插拔流程

BernardXiong, 2019-06-04:
需要先有插拔检测，SD检测的脚接个IO中断吧

zhangjun1996, 2019-06-11:
> 如题， 求SD卡热插拔流程

1，给sd卡检测引脚设置一个中断处理函数。
    1.1，如果是stm32这类的cortex-m架构芯片，那你就配置个gpio中断处理函数。
    1.2，如果是其他平台，可以是使用rt_hw_interrupt_install安装中断处理函数，使用rt_hw_interrupt_umask允许这个中断。
2，在步骤1得到的中断处理函数中调用mmcsd_change即可。

注意：处理好步骤1的操作得到中断处理函数后，最好测试下这个函数能否工作。

willianchanlovegithub, 2019-06-11:
如果是野火的那个stm32f429挑战者板子的话，是无法做到热插拔的。因为这个板子硬件上，SD卡的check pin是悬空的，没接任何东西......做不了热插拔检测。其他板子如果check pin接了的话，是可以做到热插拔检测的。

willianchanlovegithub, 2019-06-11:
当然，一直去尝试枚举SD卡，也是可以做到热插拔检测，不过十分麻烦，不如检查check pin这种方法来的直接和方便。

willianchanlovegithub, 2019-06-12:
朋友，你可以参考一下我这个pr：https://github.com/RT-Thread/rt-thread/pull/2770，在stm32f469-st-disco板子上实现了SD卡热插拔功能。这个热插拔功能是根据检测SD卡的check pin脚实现的。

closed 2019-06-14 by BernardXiong

Letian-stu, 2022-01-18:
https://club.rt-thread.org/ask/article/3276.html

Letian-stu, 2022-01-18:
发现不需要另外io也可以实现哦! https://club.rt-thread.org/ask/article/3276.html