RT-Thread/rt-thread #2776: STM32F469 中SD卡热拔插机制的BUG

whj4674672, 2019-06-14:
static void sd_mount(void *parameter)
{
    rt_uint8_t re_sd_check = 1;
    
    while (1)
    {
        rt_thread_mdelay(200);
        if(re_sd_check && !rt_pin_read(SD_CHECK_PIN))
        {
            _sdcard_mount();
        }
        if (!re_sd_check && rt_pin_read(SD_CHECK_PIN))
        {
            _sdcard_unmount();
        }
        re_sd_check = rt_pin_read(SD_CHECK_PIN);
    }
}

如果在re_sd_check = rt_pin_read(SD_CHECK_PIN);时候读取到卡插入，卡插入是低电平。
        rt_thread_mdelay(200);
        if(  0 && 1 )
        {
            _sdcard_mount();
        }
        if (  1 &&  0 )
        {
            _sdcard_unmount();
        }
        0 = rt_pin_read(SD_CHECK_PIN);

BernardXiong, 2019-06-14:
@willianchanlovegithub please take a look, thanks.

willianchanlovegithub, 2019-06-15:
@whj4674672 已修复。
```
static void sd_mount(void *parameter)
{
    rt_uint8_t re_sd_check_pin = 1;
    
    while (1)
    {
        rt_thread_mdelay(200);
        if(re_sd_check_pin && (re_sd_check_pin = rt_pin_read(SD_CHECK_PIN)) == 0)
        {
            _sdcard_mount();
        }
        
        if (!re_sd_check_pin && (re_sd_check_pin = rt_pin_read(SD_CHECK_PIN)) != 0)
        {
            _sdcard_unmount();
        }
    }
}
```

closed 2019-06-28 by BernardXiong