RT-Thread/rt-thread #2778: How to find the paths

Jack-S1008, 2019-06-14:
`#ifndef __I2C_BIT_OPS_H__
#define __I2C_BIT_OPS_H__

#ifdef __cplusplus
extern "C" {
#endif

struct rt_i2c_bit_ops
{
    void *data;            /* private data for lowlevel routines */
    void (*set_sda)(void *data, rt_int32_t state);
    void (*set_scl)(void *data, rt_int32_t state);
    rt_int32_t (*get_sda)(void *data);
    rt_int32_t (*get_scl)(void *data);

    void (*udelay)(rt_uint32_t us);

    rt_uint32_t delay_us;  /* scl and sda line delay */
    rt_uint32_t timeout;   /* in tick */
};

rt_err_t rt_i2c_bit_add_bus(struct rt_i2c_bus_device *bus,
                            const char               *bus_name);

#ifdef __cplusplus
}
#endif

#endif`
those code are in file:rt-thread/components/drivers/include/drivers/i2c-bit-ops.h
you does not include any paths,how do you find the struct rt_i2c_bus_device,rt_uint32_t and so on?
that's a question

BernardXiong, 2019-06-15:
Do you mean the path of header file?

Please use scons, which will handle compiler argument, paths and libraries etc.

zhangjun1996, 2019-06-28:
because RT-thread no use i2c-bit-ops.h independently.
RT-thread use file:rt-thread/components/drivers/include/rtdevice.h,and there is code as follow in rtdevice.h：
`#ifdef RT_USING_I2C
#include "drivers/i2c.h"
#include "drivers/i2c_dev.h"

#ifdef RT_USING_I2C_BITOPS
#include "drivers/i2c-bit-ops.h"
#endif /* RT_USING_I2C_BITOPS */
#endif /* RT_USING_I2C */`

struct rt_i2c_bus_device is defined in drivers/i2c.h

BernardXiong, 2019-06-28:
Yes, so you can include `rtdevice.h` file.

closed 2021-02-08 by mysterywolf