RT-Thread/rt-thread #2784: Hard Fault- INVPC on the first attempt of context switching

diewelt, 2019-06-17:
I'm new to arm and working on Nordic PCA10040 board which has coretex M4 processor, nrf52832. I'm trying to make rt-thread work on the board. I'm using scons and gcc tool chain. rt-thread version is recent one downloaded from git several weeks ago. I made the board work with rt-thread RTOS. But it crashed on the first attempt of context-switching. libcpu/arm/cortex-m4/context_gcc.S was used.

arm-none-eabi-gcc - 4.9.2
arm-none-eabi-ld - 2.31

This is the console output which rt-thread generated through hard fault handler. It shows the registers right after the fault.

=========================================================================================>>
     - RT -     Thread Operating System
 / | \     4.0.2 build Jun 16 2019
 2006 - 2019 Copyright by rt-thread team
psr: 0x01000000
r00: 0x00000000
r01: 0x00000000
r02: 0x00000000
r03: 0x00000000
r04: 0xdeadbeef
r05: 0xdeadbeef
r06: 0xdeadbeef
r07: 0xdeadbeef
r08: 0xdeadbeef
r09: 0xdeadbeef
r10: 0xdeadbeef
r11: 0xdeadbeef
r12: 0x00000000
 lr: 0x00008b95
 pc: 0x000098c1
hard fault on thread: timer

thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tidle0    31  ready   0x00000048 0x00000100    28%   0x00000020 000
timer      4  running 0x00000048 0x00000200    14%   0x0000000a 000
init      10  ready   0x00000044 0x00000400    06%   0x00000014 000
led       10  ready   0x00000044 0x00000400    06%   0x00000014 000
usage fault:
SCB_CFSR_UFSR:0x04 INVPC 
<<=========================================================================================

lr and pc can have odd address because there are 16 bit thumb codes. The function to be called is actually a thread entry and has one argument. But it doesn't use it and NULL is passed. So the register values look fine to me.

And I used SEGGER gdb server and gdb to debug this problem step by step. This part of the codes are part of PendSV Handler which performs context switching. The scheduler code is being called for the first time and all stack values are prepared manually, such as return address, thread entry point and stack address, and register values.

=========================================================================================>> 
pendsv_exit ()
    at /home/backup/src/arm/rt-thread/libcpu/arm/cortex-m4/context_gcc.S:154
154         MSR PRIMASK, r2
(gdb) info registers
r0             0x20000d00          536874240
r1             0x20000b84          536873860
r2             0x0                 0
r3             0x20000748          536872776
r4             0x0                 0
r5             0xdeadbeef          3735928559
r6             0xdeadbeef          3735928559
r7             0xdeadbeef          3735928559
r8             0xdeadbeef          3735928559
r9             0xdeadbeef          3735928559
r10            0xdeadbeef          3735928559
r11            0xdeadbeef          3735928559
r12            0x0                 0
sp             0x2000ffb8          0x2000ffb8
lr             0xfffffff9          4294967289
pc             0x9ed4              0x9ed4 <pendsv_exit>
xpsr           0x100000e           16777230
msp            0x2000ffb8          536936376
psp            0x20000b84          536873860
primask        0x1                 1
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
fpscr          0x0                 0
(gdb) s
156         ORR lr, lr, #0x04
(gdb) info registers
r0             0x20000d00          536874240
r1             0x20000b84          536873860
r2             0x0                 0
r3             0x20000748          536872776
r4             0x0                 0
r5             0xdeadbeef          3735928559
r6             0xdeadbeef          3735928559
r7             0xdeadbeef          3735928559
r8             0xdeadbeef          3735928559
r9             0xdeadbeef          3735928559
r10            0xdeadbeef          3735928559
r11            0xdeadbeef          3735928559
r12            0x0                 0
sp             0x2000ffb8          0x2000ffb8
lr             0xfffffff9          4294967289
pc             0x9ed8              0x9ed8 <pendsv_exit+4>
xpsr           0x100000e           16777230
msp            0x2000ffb8          536936376
psp            0x20000b84          536873860
primask        0x0                 0
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
fpscr          0x0                 0
(gdb) s
157         BX  lr
(gdb) info registers
r0             0x20000d00          536874240
r1             0x20000b84          536873860
r2             0x0                 0
r3             0x20000748          536872776
r4             0x0                 0
r5             0xdeadbeef          3735928559
r6             0xdeadbeef          3735928559
r7             0xdeadbeef          3735928559
r8             0xdeadbeef          3735928559
r9             0xdeadbeef          3735928559
r10            0xdeadbeef          3735928559
r11            0xdeadbeef          3735928559
r12            0x0                 0
sp             0x2000ffb8          0x2000ffb8
lr             0xfffffffd          4294967293
pc             0x9edc              0x9edc <pendsv_exit+8>
xpsr           0x100000e           16777230
msp            0x2000ffb8          536936376
psp            0x20000b84          536873860
primask        0x0                 0
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
fpscr          0x0                 0
(gdb) s
HardFault_Handler ()
    at /home/backup/src/arm/rt-thread/libcpu/arm/cortex-m4/context_gcc.S:221
221         MRS     r0, msp                 /* get fault context from handler. */
(gdb) info registers
r0             0x20000d00          536874240
r1             0x20000b84          536873860
r2             0x0                 0
r3             0x20000748          536872776
r4             0x0                 0
r5             0xdeadbeef          3735928559
r6             0xdeadbeef          3735928559
r7             0xdeadbeef          3735928559
r8             0xdeadbeef          3735928559
r9             0xdeadbeef          3735928559
r10            0xdeadbeef          3735928559
r11            0xdeadbeef          3735928559
r12            0x0                 0
sp             0x2000ffb8          0x2000ffb8
lr             0xfffffffd          4294967293
pc             0x9f1e              0x9f1e <HardFault_Handler>
xpsr           0x1000003           16777219
msp            0x2000ffb8          536936376
psp            0x20000b84          536873860
primask        0x0                 0
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
fpscr          0x0                 0
<<=========================================================================================

It is supposed to jump to the thread entry point through "bx lr" instruction. But it caused hard fault, INVPC.

I could not find the cause of this HARD FAULT. Could you please help me to locate the cause of this error?

Also I found out that there was "SP issue which caused hard fault on ARM cortex-M arch". I don't know about SP issue. Is my problem related to the old SP issue?

BernardXiong, 2019-06-17:
Hello, 
Which BSP do you used? This one: `rt-thread/bsp/nrf52832`

You can provide more information, for example, BSP name, tool chain (There are many GCC tool chain, such as ARM GNU GCC, native GNU GCC for arm-none-eabi-elf in Ubuntu etc) etc.

diewelt, 2019-06-17:
Yes, the board is nrf52832 and I'm using the package at **rt-thread/bsp/nrf52832**
**arm gcc - arm-none-eabi-gcc** version 4.9.2 (from https://slackbuilds.org/repository/14.2/development/arm-gcc/?search=arm-gcc)
**arm binutils - arm-none-eabi-ld** version 2.31 (https://slackbuilds.org/repository/14.2/development/arm-binutils/?search=arm-binutils)

I have no idea about the difference between arm gnu gcc or native gnu gcc. But I can tell you that I'm using slackware and I compiled the tools from the sources downloaded at slackbuilds.org.

If you need any more information, please let me know.

BernardXiong, 2019-06-17:
Mostly, we use this one:
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

You can try GCC 5/6/7. There are some conflicting in GCC 8. When you switch tool chain, please use `scons -c` firstly.

diewelt, 2019-06-17:
[led_blink.txt](https://github.com/RT-Thread/rt-thread/files/3297915/led_blink.txt)

First I upgrade gcc to 6.1.0 and binutils are not old. But the result was the same.

So I made a simple code to test the context switching.

1. thread entry
2. test code for context switching, which calls rt_hw_stack_init and rt_hw_context_switch_to

Even though I didn't use scheduler, the procedures are the same as  rt_system_scheduler_start. I placed the function context_switching_test before rt_system_scheduler_start is called.

I checked all the register values and they are correct.

I think gcc version is definitely not the cause. would you please tell me about "SP issue which caused hard fault on ARM cortex-M arch" at least?

BernardXiong, 2019-06-18:
@Zero-Free please take a look, which is GNU GCC for nRF53832.

diewelt, 2019-06-18:
I found that this problem is related to floating point option, which might affect the stack operation. When I added the option, '-mfloat-abi=soft', there is no more Usage Fault.

closed 2019-06-18 by diewelt

diewelt, 2019-07-13:
I found the final solution for hardware floating point calculation.

In contet_gcc.S, there is a macro __SOFTFP__. You shouldn't define __SOFTFP__ in compile option.

My compile and link options defined in rtconfig.py
    DEVICE = ' -mthumb -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard'
    CFLAGS = DEVICE
    CFLAGS += ' -std=c99'
    CFLAGS += ' -ffunction-sections'
    CFLAGS += ' -fdata-sections'
    AFLAGS = ' -c' + DEVICE
    **ASPFLAGS = ' -mimplicit-it=always'**
    ASPPFLAGS = ' -x assembler-with-cpp'
    ASPPFLAGS += ' -D__STARTUP_CLEAR_BSS'
    LFLAGS = DEVICE + ' -Wl,--gc-sections,-Map=rtthread-nrf52832.map,-cref,-u,Re
set_Handler -nostdlib -L nRF5_SDK_13.0.0_04a0bfd/components/toolchain/gcc -T nrf
52_xxaa.ld'

pay attention to -mimplicit-it=always. Without this option the compiler will give you some errors while assembling context_gcc.S because of the lines below.

---- snip ----
   VSTMDBEQ r1!, {d8 - d15}    /* push FPU register s16~s31 */
---- snip ----
   VLDMIANE  r1!, {d8 - d15}   /* pop FPU register s16~s31 */
---- snip ----
   BICNE   lr, lr, #0x10       /* lr &= ~(1 << 4), set FPCA. */
---- snip ----

These intructions needs some "IT NE" or "IT EQ" before themselves because they are IT conditionals. But compiler cope with them without IT istructions when given the option "-mimplicit-it=always"