RT-Thread/rt-thread #2818: About the newest version of STM32cubeMX working with the BSP template of Stm32F407ZE

arikadogenya, 2019-06-28:
when base on the template of stm32f4xx to make the BSP for a user-defined stm32f407zet using the STM32CubeMX Ver 5.2.1, only the GPIO and Uart is configured, the script tools generated a line "#define HAL_EXTI_MODULE_ENABLED" in the "stm32f4xx_hal_conf.h" which introduces an including of file "stm32f4xx_hal_exti.h" , to manage the EXTI for the MCU.
However the file "stm32f4xx_hal_exti.h" may be no longer generated by teh CubeMX and some of the content was removed to another file , which I think may be "stm32f4xx_hal_gpio.h".
Please confirm it and make the changes if if right,
Thank you and best regards.

zhangjun1996, 2019-06-28:
you can comment out `#define HAL_EXTI_MODULE_ENABLED`

SummerGGift, 2019-06-28:
We will fix the problem as soon as possible.

SummerGGift, 2019-06-29:
The reason this macro definition #define HAL_EXTI_MODULE_ENABLED appears is because you are using the 1.24 version of the F4 series HAL library.

However, the existing firmware library version is 1.21, so you can try to change the HAL library version you selected in cubemx or comment out the macro definition.

Next we will consider whether to upgrade the firmware library version of F4 series BSP to 1.24.
@arikadogenya

SummerGGift, 2019-06-29:
#2820 The PR is ready to update the F4 series firmware library to version 1.24.1.

arikadogenya, 2019-07-01:
Thank you have a nice day.
It is really hard to follow the steps of STM32CubeMx.

arikadogenya, 2019-07-01:
> you can comment out `#define HAL_EXTI_MODULE_ENABLED`

Thank you !

closed 2019-07-01 by BernardXiong