RT-Thread/rt-thread #2834: There are two warnings  when compiling a rt-thread c++ project with armcc:

alexyuoouy, 2019-07-05:
Dear madam/sir,

I want to build a c/c++ mixed project on STM32, I checked the ”Support C++ features” using the ENV tool, and added “RT_USING_LIBC” to the rtconfig.h.

[image]

 I changed main.c to main.cpp. Its contents are as follows：

```
#include <rtthread.h>
#include "Semaphore.h"
using namespace rtthread;
int main(void)
{
     /* user app entry */
    Semaphore semaphore;
    return 0;
}
```

Then I use armcc to compile the project via scons : No errors or Warnings in the compilation process.

[image]

Warning occurs in the link process : 

[image]

This is my scatter file:

```
LR_IROM1 0x08000000 0x00040000  {    ; load region size_region
  ER_IROM1 0x08000000 0x00040000  {  ; load address = execution address
   *.o (RESET, +First)
   *(InRoot$$Sections)
   .ANY (+RO)
   .ANY (+XO)
  }
  RW_IRAM1 0x20000000 0x0000C000  {  ; RW data
   .ANY (+RW +ZI)
  }
}
```

Then I found this in the ARM official documentation:

[image]

I don't know how to change this file : I added two lines in the file.

```
LR_IROM1 0x08000000 0x00040000  {    ; load region size_region
  ER_IROM1 0x08000000 0x00040000  {  ; load address = execution address
   init_aeabi.o(+RO)
   * (.init_array)
   *.o (RESET, +First)
   *(InRoot$$Sections)
   .ANY (+RO)
   .ANY (+XO)
  }
  RW_IRAM1 0x20000000 0x0000C000  {  ; RW data
   .ANY (+RW +ZI)
  }
}
```

And Two more warnings now appear : 

[image]
  
What should I do to eliminate these two warnings and call the constructor correctly when stm32 starts?

Thanks in advance for your help.

Kind regards,

WangYu

BernardXiong, 2019-07-05:
Yes, its related with object construction. You can take a try with adding a global static object.

alexyuoouy, 2019-07-09:
Thank you very much. It works.

closed 2019-07-21 by BernardXiong

Yoke365, 2021-05-28:
> Dear madam/sir,
>
> I want to build a c/c++ mixed project on STM32, I checked the ”Support C++ features” using the ENV tool, and added “RT_USING_LIBC” to the rtconfig.h.
>
> [image]
>
> I changed main.c to [...]

请问警告怎么消除的