RT-Thread/rt-thread #2838: 关于qemu-vexpress-a9 SMP模式其它核启动的触发问题。

13824125580, 2019-07-08:
rtt支持qemu-vexpress-a9 SMP启动，我跟踪了一下代码，在跟到下面的代码就跟不下去了。
rt_hw_secondary_cpu_up（void)
 30 {               
 31     extern void set_secondary_cpu_boot_address(void);
 32                 
 33     set_secondary_cpu_boot_address();
 34     __asm__ volatile ("dsb":::"memory");
 35     rt_hw_ipi_send(0, 1 << 1);                                                                                                                                                             
 36 }
疑问点在rt_hw_ipi_send这一行，对这里的两个参数，　０和1<<1不是很理解, 主要疑惑是：
１．ipi vector这里是０，但是在这之前，其它核还没有启动，并没有设置其它核ipi的中断处理handler.另一个核是如何处理这个ipi请求的呢？
２．在cpu0 bringup阶段，cpu1的状态是什么？　如何触发进入到sencondary boot entry的呢？
３．这里发送IPI设置的mask 只有１位，是否是因为只支持双核SMP?

多谢指导！

geniusgogo, 2019-07-12:
我猜：
1.其它核是在secondary_cpu_c_start里启动的并设置的中断向量，rt_hw_secondary_cpu_up里调用了set_secondary_cpu_boot_address配置了其它核的代码地址（因为此时其它核应该是休眠的），然后最后调用ipi唤醒其它核？接下来我编不下去了，继续编的话：可能其它核唤醒后就会运行之前配置的代码地址的代码secondary_cpu_start？然后在secondary_cpu_start里再调用secondary_cpu_c_start？
2.CPU0bringup阶段CPU1肯定就是WFI/WFE状态，如何触发？就是通过上面说的ipi触发？
3.这里的代码是bsp专用，bsp专用肯定就是针对bsp特定的CPU。可以看K210的BSP比较清晰。
下面我继续跟你遍一下K210的以上情况：
1.K210起来后通过CPUID决定各个核干嘛，CPU0去负责初始化，其它核负责等secondary_boot_flag就绪，也就是等CPU0调用rt_hw_secondary_cpu_up.等到之后其它核就调用secondary_cpu_c_start。在secondary_cpu_c_start里就各自核安装自己的中断向量等信息，然后启动调度器，调度自己的任务去了。
2.K210中CPU0在bringup阶段其他核在等flag。如1所述
3.K210中不需要这个ipi而是用flag代替，所以不存在这个问题。当然rt-thread的SMP是不限定核心数量的。
我编完了，请求大佬出来评判！！！

jesven, 2019-07-12:
-  (1 << 1)的问题是ipi发送的时候cpu屏蔽位，会向为1的位cpu发送ipi（1<<1 也就是cpu1)，它的中断号是0
- 关于handler的问题：这和电源管理组件是相关的，一般在对电源管理组件设定之后，执行wfi之后，cpu会进入休眠态，等待中断，当中断发生时，它并不会进入中断处理handler，而是一次复位，复位之后的代码会判断自己是不是cpu0,如果不是cpu0,会去读特定地址的值是否是0,如果是0,则休眠，否则以这个特定地址上处的数值为地址，跳转执行，从此开始了它的征程。
- 是，因为这个bsp只开启了双核，但如果想支持4核或8核，也是很简单，只要稍微改下就可以。

closed 2022-03-25 by mysterywolf