RT-Thread/rt-thread #2840: nRF52832编译错误，无法生成bin

Afantor, 2019-07-09:
首先我使用scons --target=mdk5 生成Keil工程是没有问题的，然后想直接使用scons编译，没有成功出现大量报错。如果打开keil软件进行编译，出现两个错误，下面是Keil的错误提示：
`.\build\rtthread.axf: Error: L6200E: Symbol rt_application_init multiply defined (by application.o and components.o).
.\build\rtthread.axf: Error: L6200E: Symbol rtthread_startup multiply defined (by startup.o and components.o).
Not enough information to list image symbols.
Not enough information to list load addresses in the image map.
Finished: 2 information, 0 warning and 2 error messages.
".\build\rtthread.axf" - 2 Error(s), 0 Warning(s).`

我查看过这两个API函数确实存在相同的名字在不同的文件内。

BernardXiong, 2019-07-09:
Please disable "RT_USING_USER_MAIN/The main() function as user entry function" in menuconfig.

Afantor, 2019-07-09:
> Please disable "RT_USING_USER_MAIN/The main() function as user entry function" in menuconfig.

@BernardXiong
还有问题是：我禁止了“The main() function as user entry function”后，keil的工程编译可以完成，但scons命令编译还是出现：“startups\arm\arm_startup_nrf52.s:20:28: fatal error: startup_config.h: No such file or directory”。是nRF52832不支持scons编译吗？

Afantor, 2019-07-09:
还有另外一个问题，我使用menuconfig添加了MicroPython的支持，然后使用scons --target=mdk5 生产keil工程，打开keil编译工程，出现新的报错：
`.\build\rtthread.axf: Error: L6218E: Undefined symbol __aeabi_assert (referred from argcheck.o).

.\build\rtthread.axf: Error: L6218E: Undefined symbol __signgam_addr (referred from lgamma.o).

Not enough information to list image symbols.

Not enough information to list load addresses in the image map.

Finished: 2 information, 0 warning and 2 error messages.

".\build\rtthread.axf" - 2 Error(s), 62 Warning(s).`

BernardXiong, 2019-07-10:
@SummerGGift Any comments?

SummerGGift, 2019-07-10:
我测试一下然后给出结果。

SummerGGift, 2019-07-10:
@Afantor 这个问题是因为选中了 mdk 里面的 `Use MicroMib` 选项而导致的，可以尝试关闭该选项后重新编译。

SummerGGift, 2019-07-10:
你在 issue 中提到的问题已经在 PR #2842 中修复。

closed 2021-02-06 by mysterywolf