RT-Thread/rt-thread #2891: USB CDC类 _cdc_get_line_coding函数返回野指针

gw826943555, 2019-07-24:
`
static rt_err_t _cdc_get_line_coding(udevice_t device, ureq_t setup)
{
    struct ucdc_line_coding data;
    rt_uint16_t size;

    RT_ASSERT(device != RT_NULL);
    RT_ASSERT(setup != RT_NULL);

    RT_DEBUG_LOG(RT_DEBUG_USB, ("_cdc_get_line_coding\n"));

    data.dwDTERate = 115200;
    data.bCharFormat = 0;
    data.bDataBits = 8;
    data.bParityType = 0;
    size = setup->wLength > 7 ? 7 : setup->wLength;

    rt_usbd_ep0_write(device, (void*)&data, size);

    return RT_EOK;
}
`
rt_usbd_ep0_write(device, (void*)&data, size)  中data为局部变量

BernardXiong, 2019-07-26:
@lymzzyh 这个ep0_write会是同步的还是异步的？如果是同步的，这里会存在局部变量生命周期的风险。

lymzzyh, 2019-07-29:
这里数据的发送是异步的。但是buffer的使用是同步的。底层相关的接口是同步的将数据写入到ep的缓冲区后返回所以退出函数后不再需要这个buffer

lymzzyh, 2019-07-30:
@BernardXiong 熊大看看怎么处理？

gw826943555, 2019-07-30:
> @BernardXiong 熊大看看怎么处理？

> 这里数据的发送是异步的。但是buffer的使用是同步的。底层相关的接口是同步的将数据写入到ep的缓冲区后返回所以退出函数后不再需要这个buffer，只要数据包不发生拆包，就没有生命周期的风险，但是此处代码确实写得不够合适。

目前STM32 hal USB驱动目前只是保存指针，并没有立即拷贝到ep缓冲当中，而是通过触发中断，在中断中处理。

BernardXiong, 2019-07-30:
唔，我上面说错了。应该说同步没问题，如果是异步的会有问题。

@lymzzyh 如果底层可能是异步的，那么这里需要处理好。

closed 2024-12-05 by Rbb666