RT-Thread/rt-thread #2904: 增加传感器驱动-增量型编码器

TwinMouse, 2019-07-28:
能够在传感器驱动中增加增量型AB相编码器的驱动呢，这种编码器用的很普遍，在工业控制方便，需要将定时器设定成encoder模式，读取编码器增加或者减少的数值，同时Z相出发计数值归零，做电机调速相关控制也经常用到编码器测速

BernardXiong, 2019-07-28:
RT-Thread是一个开源的社区，欢迎大家来贡献代码，驱动，软件包。

gbcwbz, 2019-08-03:
AB相编码器和硬件相关性比较大，是否能有效的抽象出来？

Guozhanxin, 2019-08-08:
> AB相编码器和硬件相关性比较大，是否能有效的抽象出来？

这个已经在做了

Guozhanxin, 2019-08-13:
这个编码器框架已经合并了，可以看一下 https://github.com/RT-Thread/rt-thread/pull/2937

TwinMouse, 2019-09-17:
rt_err_t pulse_encoder_init(struct rt_pulse_encoder_device *pulse_encoder)
{
    TIM_Encoder_InitTypeDef sConfig;
    TIM_MasterConfigTypeDef sMasterConfig;
    TIM_HandleTypeDef *tim_handler = (TIM_HandleTypeDef *)pulse_encoder->parent.user_data;

    tim_handler->Init.Prescaler = 0;
    tim_handler->Init.CounterMode = TIM_COUNTERMODE_UP;
    tim_handler->Init.Period = 0xffff;
    tim_handler->Init.ClockDivision = TIM_CLOCKDIVISION_DIV1;

    sConfig.EncoderMode = TIM_ENCODERMODE_TI12;
    sConfig.IC1Polarity = TIM_ICPOLARITY_RISING;
    sConfig.IC1Selection = TIM_ICSELECTION_DIRECTTI;
    sConfig.IC1Prescaler = TIM_ICPSC_DIV1;
    sConfig.IC1Filter = 3;
    sConfig.IC2Polarity = TIM_ICPOLARITY_RISING;
    sConfig.IC2Selection = TIM_ICSELECTION_DIRECTTI;
    sConfig.IC2Prescaler = TIM_ICPSC_DIV1;
    sConfig.IC2Filter = 3;

    if (HAL_TIM_Encoder_Init(tim_handler, &sConfig) != HAL_OK)
    {
        LOG_E("pulse_encoder init failed");
        return -RT_ERROR;
    }
选择的encoderMode是  TI1和TI2模式。这种模式下，AB两相的上升沿和下降沿都会计数，所以计数值是实际值的4倍，需要做分频。也就是第一个参数，分频值设为3，实际上是3+1=4分频。

    还有个地方需要解释一下，我刚开始的时候就是把这里的设置没搞清楚，看Polarity参数设置的是Rising Edge。这个参数的意思是在检测到上升沿的时候就触发encoder捕获AB相的值，而并不是这里设置的是上升沿就只检测AB相的上升沿，下降沿还是同样会计数的。
http://www.eemaker.com/stm32cubemx-encoder.html

balanceTWK, 2019-09-18:
> rt_err_t pulse_encoder_init(struct rt_pulse_encoder_device *pulse_encoder)
> {
> TIM_Encoder_InitTypeDef sConfig;
> TIM_MasterConfigTypeDef sMasterConfig;
> TIM_HandleTypeDef *tim_handler = [...]

有问题欢迎提 PR， 参与修改 rt-thread 代码。

closed 2021-05-23 by mysterywolf