RT-Thread/rt-thread #2939: at_client.c

Tonny-Y, 2019-08-10:
rt_size_t at_client_obj_recv()函数说明的返回值为>0或者=0，但我遇到了
[E/at.clnt] AT Client receive failed, uart device get data error(-2)
[E/at.skt] esp8266 device(esp0) receive size(4) data failed.
不知道什么原因。

BernardXiong, 2019-08-11:
可以看看这两行错误日志输出相关的代码

Lawlieta, 2019-08-22:
能否提供该问题复现环境和方式，这边好帮你复现解决问题 @Tonny-Y

Tonny-Y, 2019-09-05:
这个不是每次都会出现，之前要么上电就出问题，要么运行一段时间才出问题，这两天换了个路由器，好像再没发生过这种错误，猜测可能路由器里某些设置导致的或者是无线网络不稳定，后面再出现问题的话再来答复你哈😆，之前的一些日志也不一定有用，就不发了。

Tonny-Y, 2019-09-05:
> 
> 
> 能否提供该问题复现环境和方式，这边好帮你验证和处理问题 @Tonny-Y

这个不是每次都会出现，之前要么上电就出问题，要么运行一段时间才出问题，这两天换了个路由器，好像再没发生过这种错误，猜测可能路由器里某些设置导致的或者是无线网络不稳定，后面再出现问题的话再来答复你哈😆，之前的一些日志也不一定有用，就不发了。

Lawlieta, 2019-09-11:
最近 AT Device 软件包有针对该问题进行修复，可以试一下这个 https://github.com/RT-Thread-packages/at_device/pull/69  能否解决你的问题

closed 2021-02-06 by mysterywolf