RT-Thread/rt-thread #2941: scons --menuconfig  in  bsp/imxrt

youngday, 2019-08-10:
in /bsp/imxrt/imxrt1052-atk-commander
run `scons --menuconfig`  
error ../Libraries/Kconfig not find.
issues change Libraies to libraries in /rt-thread/bsp/imxrt

yqiu2018, 2019-08-11:
@tyustli please look at it.

tyustli, 2019-08-11:
env no this command : `scons --menuconfig`
1.if your platform is gcc:you can use command:`scons`
2.if your platform is mdk:you can use command`scons --target=mdk5`
2.if your platform is iar:you can use command:`scons --target=iar`

youngday, 2019-08-11:
env in ubuntu 

in /bsp/imxrt/imxrt1052-atk-commander
run scons --menuconfig

Libraries  need changed in to libraries

tyustli, 2019-08-11:
> in /bsp/imxrt/imxrt1052-atk-commander
> run `scons --menuconfig`
> error ../Libraries/Kconfig not find.
> issues change Libraies to libraries in /rt-thread/bsp/imxrt

when your system is linux, L and l is distinguish,to solve this error you can change L to l.
this pull request will fix this
[https://github.com/RT-Thread/rt-thread/pull/2945](url)

closed 2019-08-11 by youngday