RT-Thread/rt-thread #2959: [qemu-vexpress-a9] C++ 使用std::cout出错

djndl1, 2019-08-15:
- RTT版本：3.1.3
- 开发环境 : Debian Buster

C++ 代码
```cpp
#include <rtthread.h>
#include <string>
#include <cstdio>
#include <iostream>

void io_test(void)
{
    std::string s{"Test String"};
    
    fprintf(stdout, "%s\n", s.c_str());
    std::cout << s << std::endl;
}

MSH_CMD_EXPORT(io_test, C++ test I/O);
```

错误信息
```
WARNING: Image format was not specified for 'sd.bin' and probing guessed raw.                          
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.                               
pulseaudio: set_sink_input_volume() failed
pulseaudio: Reason: Invalid argument
pulseaudio: set_sink_input_mute() failed
pulseaudio: Reason: Invalid argument

 \ | /
- RT -     Thread Operating System
 / | \     3.1.3 build Aug 16 2019
 2006 - 2019 Copyright by rt-thread team
lwIP-2.0.2 initialized!
[I/sal.skt] Socket Abstraction Layer initialize success.                                               
[I/SDIO] SD card capacity 65536 KB.
[I/SDIO] switching card to high speed failed!
file system initialization done!
hello rt-thread
msh />io
io_test
msh />io_test
Test String
data abort:Execption:
r00:0x600f6618 r01:0x600e605c r02:0x00000000 r03:0x00000000                                            
r04:0x600f34c4 r05:0x600f6618 r06:0xdeadbeef r07:0x0000000b                                            
r08:0xdeadbeef r09:0xdeadbeef r10:0xdeadbeef
fp :0x600f6654 ip :0x01010101
sp :0x600f6608 lr :0x60099411 pc :0x60099274
cpsr:0x40000033
thread   pri  status      sp     stack size max used left tick  error                                  
-------- ---  ------- ---------- ----------  ------  ---------- ---                                    
tshell    20  ready   0x00000184 0x00001000    21%   0x00000003 000                                    
aio       16  suspend 0x00000084 0x00000800    06%   0x0000000a 000                                    
sys_work  23  suspend 0x00000080 0x00000800    49%   0x00000005 000                                    
mmcsd_de  22  suspend 0x000000b0 0x00000400    53%   0x0000000e 000                                    
tcpip     10  suspend 0x000000ec 0x00000400    83%   0x0000000a 000                                    
etx       12  suspend 0x000000ac 0x00000400    16%   0x0000000f 000                                    
erx       12  suspend 0x000000b0 0x00000400    57%   0x0000000f 000                                    
tidle     31  ready   0x0000005c 0x00000400    14%   0x00000004 000                                    
timer      4  suspend 0x0000007c 0x00000400    16%   0x00000008 000                                    
shutdown...
(0) assertion failed at function:rt_hw_cpu_shutdown, line number:79                                    
```

然后就在一个死循环里面打转。

BernardXiong, 2019-08-16:
这个应该是一些std c++的移植未对接

djndl1, 2019-08-16:
顺便问一下，有没有可能支持C++11的多线程，既然pthead都有了，外面封装一层应该不难吧。

BernardXiong, 2019-08-26:
@tyustli 上面C++的问题可否帮忙看看？

@djndl1 
C++11多线程，估计需要先把整体C++11相关的移植弄清楚后才来考虑。或者，您这边可以是否先行推进下去？

tyustli, 2019-08-26:
> @tyustli 上面C++的问题可否帮忙看看？
> 
> @djndl1
> C++11多线程，估计需要先把整体C++11相关的移植弄清楚后才来考虑。或者，您这边可以是否先行推进下去？

好的

tyustli, 2019-08-26:
@djndl1 你好，使用 C++ 之前，建议你先看一下 rt-thread\components\cplusplus 文件夹下面的 READMD.md 文件，里面有对 C++ 使用的说明。
下面是这个 link.lds 文件是我改的，在 qemu-vexpress-a9 上运行你提供的 io_test 是可以测试成功的：
`OUTPUT_FORMAT("elf32-littlearm", "elf32-littlearm", "elf32-littlearm")
OUTPUT_ARCH(arm)
SECTIONS
{
    . = 0x60010000;

    __text_start = .;
    .text :
    {
        *(.vectors)
        *(.text)
        *(.text.*)

        /* section information for utest */
        . = ALIGN(4);
        __rt_utest_tc_tab_start = .;
        KEEP(*(UtestTcTab))
        __rt_utest_tc_tab_end = .;
        
        /* section information for finsh shell */
        . = ALIGN(4);
        __fsymtab_start = .;
        KEEP(*(FSymTab))
        __fsymtab_end = .;
        . = ALIGN(4);
        __vsymtab_start = .;
        KEEP(*(VSymTab))
        __vsymtab_end = .;
        . = ALIGN(4);

        /* section information for modules */
        . = ALIGN(4);
        __rtmsymtab_start = .;
        KEEP(*(RTMSymTab))
        __rtmsymtab_end = .;

        /* section information for initialization */
        . = ALIGN(4);
        __rt_init_start = .;
        KEEP(*(SORT(.rti_fn*)))
        __rt_init_end = .;

        . = ALIGN(4);

        PROVIDE(__ctors_start__ = .);
        KEEP (*(SORT(.init_array.*)))
        KEEP (*(.init_array))
        PROVIDE(__ctors_end__ = .);

        . = ALIGN(4);

    } =0
    __text_end = .;

    __exidx_start = .;
    .ARM.exidx   : 
    {
        *(.ARM.exidx* .gnu.linkonce.armexidx.*) 
    }
    __exidx_end = .;

    __rodata_start = .;
    .rodata   : { *(.rodata) *(.rodata.*) }
    __rodata_end = .;

    __data_start = .;
    . = ALIGN(4);
    .data :
    {
        *(.data)
        *(.data.*)
        PROVIDE(__dtors_start__ = .);
        KEEP(*(SORT(.dtors.*)))
        KEEP(*(.dtors))
        PROVIDE(__dtors_end__ = .);
        
        . = ALIGN(4);
    }
    __data_end = .;

    . = ALIGN(4);
    __bss_start = __data_end;
    .bss       :
    {
    *(.bss)
    *(.bss.*)
    *(COMMON)
    . = ALIGN(4);
    }
    . = ALIGN(4);
    __bss_end = .;

    /* Stabs debugging sections.  */
    .stab 0 : { *(.stab) }
    .stabstr 0 : { *(.stabstr) }
    .stab.excl 0 : { *(.stab.excl) }
    .stab.exclstr 0 : { *(.stab.exclstr) }
    .stab.index 0 : { *(.stab.index) }
    .stab.indexstr 0 : { *(.stab.indexstr) }
    .comment 0 : { *(.comment) }

    _end = .;
}
`

BernardXiong, 2019-08-27:
@tyustli 链接脚本有什么不同吗？

tyustli, 2019-08-27:
> @tyustli 链接脚本有什么不同吗？

这个是以前 C++ 的段
`
    . = ALIGN(4);
    .ctors :
    {
        PROVIDE(__ctors_start__ = .);
        KEEP(*(SORT(.ctors.*)))
        KEEP(*(.ctors))
        PROVIDE(__ctors_end__ = .);
    }

    .dtors :
    {
        PROVIDE(__dtors_start__ = .);
        KEEP(*(SORT(.dtors.*)))
        KEEP(*(.dtors))
        PROVIDE(__dtors_end__ = .);
    }
`
 这个是改完之后 C++ 的段
`.text :
{
   ......
    . = ALIGN(4);

    PROVIDE(__ctors_start__ = .);
    KEEP (*(SORT(.init_array.*)))
    KEEP (*(.init_array))
    PROVIDE(__ctors_end__ = .);

    . = ALIGN(4);

} =0

.data :
{
    *(.data)
    *(.data.*)
    PROVIDE(__dtors_start__ = .);
    KEEP(*(SORT(.dtors.*)))
    KEEP(*(.dtors))
    PROVIDE(__dtors_end__ = .);
    
    . = ALIGN(4);
}

`

djndl1, 2019-08-27:
@tyustli 改了之后就没问题了

@BernardXiong C++11 线程这块，在类unix上面我看了一下libc++，基本就是围着Pthread封装了一圈，我看之后有时间弄一下吧。

closed 2019-08-27 by djndl1

oska874, 2019-12-03:
修改了lds 就好了，为什么不合并进代码呢？