RT-Thread/rt-thread #2960: find a bug in drv_sdram.c

feiliuzhixia1000m, 2019-08-16:
Hello , I use RT1052 Fire Pro + RT_Thread, I want to Move the Interrupt vector to SDRAM 0x80000000, so I write a move procedue in the function ConfigMpu(), but in  drv_sdram.c:line 72
I find the function rt_memheap_init() init the entire sdram ,so my new interrupt vector is corrupted, an i think maybe other data have been corrupted too.
I think rt_memheap_init() maybe just initialize the area that i have assigned in the .scf file.

Thanks

feiliuzhixia1000m, 2019-08-16:
I find that in board.c rt_hw_board_init() calls rt_system_heap_init(), which calls  rt_memheap_init() to initialzie the memheap , so i comment the rt_memheap_init() in drv_sdram.c:line 72,and it works.

BernardXiong, 2019-08-16:
Not a bug, just adjust it according to your setting.

closed 2019-08-16 by BernardXiong