RT-Thread/rt-thread #2973: hard fault

feiliuzhixia1000m, 2019-08-20:
I use RT1052-fire-pro and rt-thread , when i usd fal module, i input "fal xxx" in the console,  and a hard fault comes out, I debug and found that the problem come from strcmp(), so I change strcmp() with rt_strcmp() and it works.
Today i import easyflash and dfs, and i input command and a new hard fault comes, i change strcmp() with rt_strcmp and i am not lucky any more!

msh />rebg
psr: 0x61005c00
r00: 0x80405bac
r01: 0x80404a3a
r02: 0x00000000
r03: 0x00000002
r04: 0x00000004
r05: 0x00000004
r06: 0x80404a3a
r07: 0x80405bac
r08: 0x00000004
r09: 0xdeadbeef
r10: 0xdeadbeef
r11: 0xdeadbeef
r12: 0x00000000
 lr: 0x80011947
 pc: 0x80000ad8
hard fault on thread: tshell

thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tshell    20  running 0x00000140 0x00001000    09%   0x00000009 000
tidle0    31  ready   0x00000048 0x00000100    34%   0x00000017 000
main      10  suspend 0x00000088 0x00001000    08%   0x00000005 000
usage fault:
SCB_CFSR_UFSR:0x100 UNALIGNED

feiliuzhixia1000m, 2019-08-20:
I debuged and found the problem is in mesh_file.c:80 , memcpy() . the program runs to here and hard fault comes.

feiliuzhixia1000m, 2019-08-20:
I replace the memcpy() with rt_memcpy(). It works again .

But can anybody tell me why?

BernardXiong, 2019-08-20:
The rt_memcpy/rt_strcmp is more safty for memory operations, for example memory alignment, cross memory block etc.

feiliuzhixia1000m, 2019-08-21:
I see, thank you

closed 2019-08-21 by BernardXiong