RT-Thread/rt-thread #3008: Signal is not supported in the RT-Thread V4.0.x

zcj20080882, 2019-08-27:
At present，signal is not supported in the RT-Thread V4.0.x，and when to support the signal？

BernardXiong, 2019-08-27:
It's supported.

closed 2019-08-27 by BernardXiong

zcj20080882, 2019-09-02:
但是我用官网上的例子测试，发现无法接收信号哦。

**1. 测试代码**

```
#define THREAD_PRIORITY         25
#define THREAD_STACK_SIZE       512
#define THREAD_TIMESLICE        5

static rt_thread_t tid1 = RT_NULL;

/* 线程 1 的信号处理函数 */
void thread1_signal_handler(int sig)
{
    rt_kprintf("thread1 received signal %d\n", sig);
}

/* 线程 1 的入口函数 */
static void thread1_entry(void *parameter)
{
    int cnt = 0;

    /* 安装信号 */
    rt_signal_install(SIGUSR1, thread1_signal_handler);
    rt_signal_unmask(SIGUSR1);

    /* 运行 10 次 */
    while (cnt < 10)
    {
        /* 线程 1 采用低优先级运行，一直打印计数值 */
        rt_kprintf("thread1 count : %d\n", cnt);

        cnt++;
        rt_thread_mdelay(100);
    }
}

/* 信号示例的初始化 */
int signal_sample(void)
{
    /* 创建线程 1 */
    tid1 = rt_thread_create("thread1",
                            thread1_entry, RT_NULL,
                            THREAD_STACK_SIZE,
                            THREAD_PRIORITY, THREAD_TIMESLICE);

    if (tid1 != RT_NULL)
        rt_thread_startup(tid1);

    rt_thread_mdelay(300);

    /* 发送信号 SIGUSR1 给线程 1 */
    rt_thread_kill(tid1, SIGUSR1);

    return 0;
}

/* 导出到 msh 命令列表中 */
MSH_CMD_EXPORT(signal_sample, signal sample);
```

**2. 配置**

[image]

**3. 测试平台/环境**

stm32l452-st-nucleo

**4. RT-Thread版本**

V4.0.2

**5. 测试结果**

[image]

从运行结构看，任务“thread1”并没有收到 msh任务发送给他的信号。

请问，是因为配置不对吗？