RT-Thread/rt-thread #3014: scons 编译额外库 Fatal error: L3900U:

liliangego, 2019-08-27:
添加库如下：
[image]

lib库(keil_lib_power_cm33_core0.lib)放在目录：libraries\LPCXpresso55S69\devices\LPC55S69\arm\libs

用scons编译是报错：
Fatal error: L3900U: Unrecognized option '-Llibraries\LPCXpresso55S69\devices\LPC55S69\arm\libs'.

[image]

BernardXiong, 2019-08-28:
Keil支持得可能有些问题，Keil的模式应该不是`-L`参数

liliangego, 2019-08-28:
> 
> 
> Keil支持得可能有些问题，Keil的模式应该不是`-L`参数

我看到building.py中使用的 armcc，我用的是armclang，这里需要适配下？

liliangego, 2019-08-28:
> 
> 
> Keil支持得可能有些问题，Keil的模式应该不是`-L`参数

在building.py中适配了armclang后，这步可以通过了，
[image]
但是有报错了？
Error: L6242E: Cannot link object fsl_power_lib.o as its attributes are incompatible with the image attributes.

在生产的KEIL工程中编译能通过呀？？？

liliangego, 2019-08-28:
> 
> 
> > Keil支持得可能有些问题，Keil的模式应该不是`-L`参数
> 
> 在building.py中适配了armclang后，这步可以通过了，
> [image]
> 但是有报错了？
> Error: L6242E: Cannot link object fsl_power_lib.o as its attributes are incompatible with the image attributes.
> 
> 在生产的KEIL工程中编译能通过呀？？？

[image]
应该是这个原因引起的。

closed 2019-08-29 by liliangego