RT-Thread/rt-thread #3018: osx环境下编译错误

zhjc2014, 2019-08-29:
提示找不到  "sh: arm-none-eabi-gcc: command not found"
```
zhandeMacBook-Pro:stm32f407-zqwl-v1.1 zhanjuncheng$ scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/applications/main.o
sh: arm-none-eabi-gcc: command not found
scons: *** [build/applications/main.o] Error 127
scons: building terminated because of errors.
```
实际上已经配置了路径
```
 cross_tool provides the cross compiler
# EXEC_PATH is the compiler execute path, for example, CodeSourcery, Keil MDK, IAR
if  CROSS_TOOL == 'gcc':
    PLATFORM    = 'gcc'
    # EXEC_PATH   = r'C:\Users\XXYYZZ'
    EXEC_PATH   = '/Users/zhanjuncheng/Downloads/gcc-arm-none-eabi-8-2018-q4-major/arm-none-eabi/bin'
elif CROSS_TOOL == 'keil':
    PLATFORM    = 'armcc'
    EXEC_PATH   = r'C:/Keil_v5'
```

SummerGGift, 2019-08-29:
arm-none-eabi 这个目录看起来不太正常，是不是前面少了 gcc- 的前缀?

SummerGGift, 2019-08-29:
https://www.rt-thread.org/qa/thread-421568-1-1.html 配置过程可以参考这个帖子

zhjc2014, 2019-08-29:
gcc-arm-none-eabi-7-2017-q4-major 这个版本可以

zhjc2014, 2019-08-29:
gcc-arm-none-eabi-8-2018-q4-major 这个版本不行

SummerGGift, 2019-08-29:
我自己用的是这个版本 gcc-arm-none-eabi-5_4-2016q3-20160926-mac.tar.bz2
 
因为现在 rt-thread 的 ci 和 env 工具用的都是这个版本，分享给你吧。

链接：https://pan.baidu.com/s/1s0M5eM4aOVpq5E3bjXnENA 
提取码：rj7w

closed 2019-08-30 by BernardXiong