RT-Thread/rt-thread #3046: 在SPI中的SConscript加SFUD库附加的c99 flag语句判断引起MDK5.28 Compiler V6.12编译出错

charlown, 2019-09-07:
使用mdk5.28，编译stm32f401rc
添加SFUD后，编译器选择default version 5，c/c++面板中默认选择C99，编译没有问题；选择default version 6，c/c++面板默认选择c99，编译会报" ArmClang: error: unsupported option '--c99' "这样的错误。在components\drivers\spi\SConscript发现去掉
if rtconfig.CROSS_TOOL == 'gcc':
        LOCAL_CCFLAGS += ' -std=c99'
    elif rtconfig.CROSS_TOOL == 'keil':
        LOCAL_CCFLAGS += ' --c99'“
这语句，scons重新生成再编译，version 5 or 6 都没问题。
---------------------------------------------------------------------------------
IDE-Version:
μVision V5.28.0.0
Copyright (C) 2019 ARM Ltd and ARM Germany GmbH. All rights reserved.

Tool Version Numbers:
Toolchain:        MDK-ARM Professional  Version: 5.28.0.0
C Compiler:         Armcc.exe        V5.06 update 6 (build 750)
Assembler:          Armasm.exe        V5.06 update 6 (build 750)
Linker/Locator:     ArmLink.exe        V5.06 update 6 (build 750)
Library Manager:    ArmAr.exe        V5.06 update 6 (build 750)
Hex Converter:      FromElf.exe        V5.06 update 6 (build 750)
CPU DLL:               SARMCM3.DLL          V5.28.0.0
Dialog DLL:         TCM.DLL              V1.36.2.0
Target DLL:             UL2CM3.DLL           V1.162.16.0
Dialog DLL:         TCM.DLL              V1.36.2.0

Tool Version Numbers:
Toolchain:        MDK-ARM Professional  Version: 5.28.0.0

C Compiler:         ArmClang.exe        V6.12
Assembler:          Armasm.exe        V6.12
Linker/Locator:     ArmLink.exe        V6.12
Library Manager:    ArmAr.exe        V6.12
Hex Converter:      FromElf.exe        V6.12
CPU DLL:               SARMCM3.DLL          V5.28.0.0
Dialog DLL:         TCM.DLL              V1.36.2.0
Target DLL:             Segger\JL2CM3.dll      V2.99.36.0
Dialog DLL:         TCM.DLL              V1.36.2.0

BernardXiong, 2019-09-10:
是的，AC6后命令行参数改了。c99是纠结的语法，原来RTT平台中能不用就不使用的

xiaozhigithub, 2020-01-16:
正找不到这个--c99在哪里定义呢，点个赞

shenyehui, 2020-10-02:
components \ drivers \ spi \ SConscrip   请问这个在什么地方

xiaozhigithub, 2020-10-02:
用搜索功能搜索一下报错相关关键字，找到sconscript中引起报错的部分，干掉他

mysterywolf, 2022-01-20:
这个地方确实有问题 应该根据platform来判断 不是根据CROSS_TOOL来判断

mysterywolf, 2022-01-20:
https://github.com/RT-Thread/rt-thread/pull/4547/commits/791527caca746633fb28119a7c8987e297fe27df
已经在此处修复了
修复时间：29 Mar 2021

closed 2022-01-20 by mysterywolf