RT-Thread/rt-thread #3085: 建议增加对2G/3G/4G模块的协议栈原生支持

xfan1024, 2019-09-18:
hello @BernardXiong 

2G/3G/4G模块是物联网中很常用的组件，目前RT-Thread中似乎缺少使用协议栈对接2G/3G/4G模块的部分。所以我做了这一部分.

希望可以考虑将此作为设备驱动纳入RT-thread系统中，后续靠社区的力量来支持更多的不同类型的设备。

_Originally posted by @xfan1024 in https://github.com/RT-Thread/rt-thread/pull/3082#issuecomment-532497003_

armink, 2019-09-18:
因为串口框架本身是一个比较基础的 **片上外设** 的设备驱动框架，你这个好像更像是基于串口接口的片外模块设备？而这类模块性质对应的软件组件，灵活性非常大，放在 **[软件包](http://packages.rt-thread.org)** 其实更加合适，也同样能方便社区力量来共同维护。

另外，我们最近也在处理通过 ppp/rndis 接入蜂窝模组的工作，近期会增加 ppp 软件包，里面会同时支持串口或 usb 方式的接口，还会完善 usb 协议栈，支持 rndis 主机。

也欢迎你也能一起加入进来，两边的代码看下是否能够合并，这里有个 [QQ 群](https://jq.qq.com/?_wv=1027&k=59xJD2b) ，可以加入进来一起交流

最后，也感谢你的建议和贡献，欢迎多多交流

xfan1024, 2019-09-18:
> 因为串口框架本身是一个比较基础的 **片上外设** 的设备驱动框架，你这个好像更像是基于串口接口的片外模块设备？而这类模块性质对应的软件组件，灵活性非常大，放在 **[软件包](http://packages.rt-thread.org)** 其实更加合适，也同样能方便社区力量来共同维护。
>
> 另外，我们最近也在处理通过 ppp/rndis 接入蜂窝模组的工作，近期会增加 ppp [...]
你好 @armink 

`modem设备驱动`之于`串口框架`的关系实际上等同于`SPI FLASH设备驱动`之于 `SPI框架`的关系。的确可以放软件包里去做，不过还是需要在`struct rt_serial_device`中增加一个user_data的成员。

我比较建议放到系统内的原因是对于物联网系统而言，2G/3G/4G网络模块的常用程度其实是比较高的。就像是SPI FLASH驱动一样`components/drivers/spi/spi_flash_sfud.c`，这个PR主要也是借鉴了SPI框架下的一些做法。

我已经申请加群了，麻烦通过一下。后续如果需要做成软件包周末的时候我可以做一下。

closed 2019-09-18 by xfan1024