RT-Thread/rt-thread #3113: pthread_create 异常分支有内存泄漏

qinpan1003, 2019-09-30:
pthread_create 
--> stack = (void *)rt_malloc(ptd->attr.stacksize);

此时 stack 还未赋值到 ptd->tid->stack_addr，以下两个地方失败会导致 stack 不被释放
--> ptd->tid = (rt_thread_t) rt_malloc(sizeof(struct rt_thread));          
--> ptd->joinable_sem = rt_sem_create(name, 0, RT_IPC_FLAG_FIFO);

并且由于 ptd->tid->stack_addr 未赋值
  _pthread_data_destroy -->  rt_free(ptd->tid->stack_addr); 也是非法操作

murphyzhao, 2019-10-03:
> pthread_create
> --> stack = (void *)rt_malloc(ptd->attr.stacksize);
>
> 此时 stack 还未赋值到 ptd->tid->stack_addr，以下两个地方失败会导致 stack 不被释放
> --> ptd->tid = (rt_thread_t) rt_malloc(sizeof(struct rt_thread));
> --> [...]

看了下，`pthread_create`  在异常的情况下确实有内存问题

BernardXiong, 2019-10-11:
fix with #3124 , thank for your contribution.

closed 2019-10-11 by BernardXiong