RT-Thread/rt-thread #3131: [bsp][stm32] 建议增加 uart DMA 接收buffsize自定义机制

a1012112796, 2019-10-12:
各大佬好，最近在使用stm32 uart DMA接收时遇到了一个问题，就是不同uart 的DMA接收区大小不可自定义,这对实际应用十分不利。仔细分析了一下相关源代码，首先在serial.c中，规定bufsz不可被动态改变(具体位置:serial.c 957行) ，这一点可以理解，因为DMA 缓存区需要分配内存空间，且与底层硬件配置有关，无法动态改变。而在驱动代码中，有使用了一个简单的for循环把每个uart dma bufsz都初始化为了RT_SERIAL_RB_BUFSZ。(具体位置:  drv_usart.c int rt_hw_usart_init(void); 第968行左右)。于是就造成了所有uart的DMA 接收 bufsz  只能被设为相同大小。但在实际应用中可能需要不同UART DMA接收设置为不同的缓存区大小。 :cry:
我觉可以利用宏定义机制，设定不同UART的DMA bufsz大小，然后在rt_hw_usart_init(void);函数中，根据宏定义分别对不同uart的DMA 接收bufsz 的值。这样应该会更合理一些。
我因为对RTT的内核代码不是很熟悉，所以没敢贸然动手改。此建议仅供各位大佬参考，如有错误还请各位大佬海涵，我只是一个小白。谢谢！

armink, 2019-10-15:
虽然串口的 buf 不能动态改变，但应该可以在 open 前，提前通过 control 函数指定好 buf_size ，这样每个设备可以有不同的 buf_szie 。你可以试试

a1012112796, 2019-10-18:
@armink 大佬您好,首先感谢您的回答，但是根据serial.c中相关代码:
[image]
该方法无法实现。
1.第962行规定，在打开设备前，无法对设备做control 的操作。
2.从逻辑上看,在 int rt_hw_usart_init(void);之后对bufsize进行修改，都属于动态改变bufsize大小，故是无法实现的。
可能大佬误会我的意思了，我不是想实现动态修改bufsize的大小在,这并没有太大意义，只是想实现不同uart在初始化时可以设定为不同的这，而不是全部为RT_SERIAL_RB_BUFSZ。
谢谢！

armink, 2019-10-18:
那是不是把 962 行的限制取消就好了，不太清楚原始代码作者为啥要加个限制

a1012112796, 2019-10-18:
@armink
今天分析了一下代码：
seial.c中 rt_serial_open函数实现了buf_size 内存分配代码如下
```
static rt_err_t rt_serial_open(struct rt_device *dev, rt_uint16_t oflag)
{
   ....................
    /* initialize the Rx/Tx structure according to open flag */
    if (serial->serial_rx == RT_NULL)
    { 
        if (oflag & RT_DEVICE_FLAG_INT_RX)
        {
            struct rt_serial_rx_fifo* rx_fifo;
      .......................
        else if (oflag & RT_DEVICE_FLAG_DMA_RX)
        {
            if (serial->config.bufsz == 0) {
                struct rt_serial_rx_dma* rx_dma;
               /**********************************/
                rx_dma = (struct rt_serial_rx_dma*) rt_malloc (sizeof(struct rt_serial_rx_dma));
                RT_ASSERT(rx_dma != RT_NULL);
                rx_dma->activated = RT_FALSE;

                serial->serial_rx = rx_dma;
            } else {
                struct rt_serial_rx_fifo* rx_fifo;

                rx_fifo = (struct rt_serial_rx_fifo*) rt_malloc (sizeof(struct rt_serial_rx_fifo) +
                    serial->config.bufsz);
                RT_ASSERT(rx_fifo != RT_NULL);
                rx_fifo->buffer = (rt_uint8_t*) (rx_fifo + 1);
                rt_memset(rx_fifo->buffer, 0, serial->config.bufsz);
                rx_fifo->put_index = 0;
                rx_fifo->get_index = 0;
                rx_fifo->is_full = RT_FALSE;
                serial->serial_rx = rx_fifo;
                /* configure fifo address and length to low level device */
                serial->ops->control(serial, RT_DEVICE_CTRL_CONFIG, (void *) RT_DEVICE_FLAG_DMA_RX);
            }
            dev->open_flag |= RT_DEVICE_FLAG_DMA_RX;
        }
#endif /* RT_SERIAL_USING_DMA */
        else
        {
            serial->serial_rx = RT_NULL;
        }
    }
    else
    {
        if (oflag & RT_DEVICE_FLAG_INT_RX)
            dev->open_flag |= RT_DEVICE_FLAG_INT_RX;
#ifdef RT_SERIAL_USING_DMA
        else if (oflag & RT_DEVICE_FLAG_DMA_RX)
            dev->open_flag |= RT_DEVICE_FLAG_DMA_RX;
#endif /* RT_SERIAL_USING_DMA */  
    }
```

1. 但从这段代码看在第一次open前通过control函数修改buffer_size应该是可行的，但还得考虑底层能否实现，也就是
···
/* configure fifo address and length to low level device */
                serial->ops->control(serial, RT_DEVICE_CTRL_CONFIG, (void *) RT_DEVICE_FLAG_DMA_RX);
···
，对stm32是没问题的。但第二次以后就有内存越界的可能了。这应该是第955行的原因吧。
2. 至于 962 行的限制，我没有找到明确的原因，但觉得应该有原因，毕竟是写进文档中心了的内容。不过文档中心确实提到了缓存区修改：
[image]
但并未说明如如何实现 调大缓冲区 😢
3. 我觉得还是在在rt_hw_usart_init(void);函数实现对buffer_size的自定义更加安全一些。
一点我的个人意见。

a1012112796, 2019-10-18:
或者可以在
```
else
    {
        if (oflag & RT_DEVICE_FLAG_INT_RX)
            dev->open_flag |= RT_DEVICE_FLAG_INT_RX;
#ifdef RT_SERIAL_USING_DMA
        else if (oflag & RT_DEVICE_FLAG_DMA_RX)
            dev->open_flag |= RT_DEVICE_FLAG_DMA_RX;
#endif /* RT_SERIAL_USING_DMA */  
    }
```
处实现当缓存区大小变化时，重新分配内存。但感觉可能会在别的地方出问题，并且实现存在困难。

a1012112796, 2019-10-18:
并且个人不是很理解为什么缓存区内存分配要在第一次打开设备时进行，而不是配置的时候？

BernardXiong, 2019-10-21:
像是stm32驱动的问题，对应的驱动单元测试需要考虑更多了

a1012112796, 2019-10-21:
各位大佬好，今天我抱着试一试的态度,在开发板上尝试了在open前control,结果竟然是可行的。在单步调试之后，突然发现，由于之前是直接看的文档中心，先入为主的接受了“打开设备后才可修改串口配置参数”的观点。今天我发现似乎是错的。serial.c中相关代码如下:
[image]
之前理解有误，其实在第961行就已经完成了配置参数的修改，只是，如果没有打开设备，则不会立即只执行配置，而是等到打开设备时，才具体执行。
测试代码如下：改编自文档中心例程，注释未更改
```

/*
 * 程序清单：这是一个串口设备 DMA 接收使用例程
 * 例程导出了 uart_dma_sample 命令到控制终端
 * 命令调用格式：uart_dma_sample uart3
 * 命令解释：命令第二个参数是要使用的串口设备名称，为空则使用默认的串口设备
 * 程序功能：通过串口输出字符串"hello RT-Thread!"，并通过串口输出接收到的数据，然后打印接收到的数据。
*/

#include <rtthread.h>
#include <rtdevice.h>
#include <stdio.h>

#define SAMPLE_UART_NAME       "uart6"      /* 串口设备名称 */

/* 串口接收消息结构*/
struct rx_msg
{
    rt_device_t dev;
    rt_size_t size;
};
/* 串口设备句柄 */
static rt_device_t serial;
/* 消息队列控制块 */
static struct rt_messagequeue rx_mq;

/* 接收数据回调函数 */
static rt_err_t uart_input(rt_device_t dev, rt_size_t size)
{
    struct rx_msg msg;
    rt_err_t result;
    msg.dev = dev;
    msg.size = size;

    result = rt_mq_send(&rx_mq, &msg, sizeof(msg));
    if ( result == -RT_EFULL)
    {
        /* 消息队列满 */
        rt_kprintf("message queue full！\n");
    }
    return result;
}

static void serial_thread_entry(void *parameter)
{
    struct rx_msg msg;
    rt_err_t result;
    rt_uint32_t rx_length;
    static char rx_buffer[RT_SERIAL_RB_BUFSZ + 1];
    char tmp[30] = {0};

    while (1)
    {
        rt_memset(&msg, 0, sizeof(msg));
        /* 从消息队列中读取消息*/
        result = rt_mq_recv(&rx_mq, &msg, sizeof(msg), RT_WAITING_FOREVER);
        if (result == RT_EOK)
        {
            /* 从串口读取数据*/
            rx_length = rt_device_read(msg.dev, 0, rx_buffer, msg.size);
            rx_buffer[rx_length] = '\0';
            /* 通过串口设备 serial 输出读取到的消息 */
            rt_device_write(serial, 0, rx_buffer, rx_length);
            sprintf(tmp,"\n(   lenth = %d   )\n",rx_length);
            rt_device_write(serial, 0, tmp, sizeof(tmp));
        }
    }
}

static int uart_dma_sample(void)
{
    rt_err_t ret = RT_EOK;
    static char msg_pool[256];
    char str[] = "hello RT-Thread!\r\n";
    struct serial_configure config = RT_SERIAL_CONFIG_DEFAULT; /* 配置参数 */

    /* 查找串口设备 */
    serial = rt_device_find(SAMPLE_UART_NAME);
    if (!serial)
    {
        rt_kprintf("find %s failed!\n", SAMPLE_UART_NAME);
        return RT_ERROR;
    }

    /* 初始化消息队列 */
    rt_mq_init(&rx_mq, "rx_mq",
               msg_pool,                 /* 存放消息的缓冲区 */
               sizeof(struct rx_msg),    /* 一条消息的最大长度 */
               sizeof(msg_pool),         /* 存放消息的缓冲区大小 */
               RT_IPC_FLAG_FIFO);        /* 如果有多个线程等待，按照先来先得到的方法分配消息 */
    
    config.baud_rate = BAUD_RATE_9600;   /*** 尝试修改波特率 */
    config.data_bits = DATA_BITS_8;
    config.stop_bits = STOP_BITS_1;
    config.bufsz     = 10;                             /** 尝试修改bufsz */
    config.parity = PARITY_NONE;
    /* 打开设备后才可修改串口配置参数 */ /** 尝试在open前执行control */
    ret = rt_device_control(serial, RT_DEVICE_CTRL_CONFIG, &config);
    if(RT_EOK != ret)
    {
        RT_ASSERT(0);
    }
    /* 以 DMA 接收及轮询发送方式打开串口设备 */
    rt_device_open(serial, RT_DEVICE_FLAG_DMA_RX);
    /* 设置接收回调函数 */
    rt_device_set_rx_indicate(serial, uart_input);
    /* 发送字符串 */
    rt_device_write(serial, 0, str, (sizeof(str) - 1));

    /* 创建 serial 线程 */
    rt_thread_t thread = rt_thread_create("serial", serial_thread_entry, RT_NULL, 1024, 25, 10);
    /* 创建成功则启动线程 */
    if (thread != RT_NULL)
    {
        rt_thread_startup(thread);
    }
    else  
    {
        ret = RT_ERROR;
    }

    return ret;
}
/* 导出到 msh 命令列表中 */
INIT_APP_EXPORT(uart_dma_sample);
```
测试平台：stm32f427_robomaster_a板 MDK5
结果：
[image]
如图可知，波特率和bufsz确实被修改了。
所以，我认为文档中如下内容有误，请验证，谢谢。
[image]

@BernardXiong

armink, 2019-10-29:
@yangjie11 找时间修正下文档中的问题吧

yangjie11, 2019-10-29:
好的，我看下

yangjie11, 2019-11-01:
文档已修正：https://www.rt-thread.org/document/site/programming-manual/device/uart/uart/#_4

closed 2019-11-02 by a1012112796