RT-Thread/rt-thread #3133: 使用 gcc 8.3.1编译有错误

easonxlee, 2019-10-14:
使用gcc 8.3.1编译时有错误。

gcc version 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
error:
\env\tools\gnu_gcc\arm_gcc\mingw\arm-none-eabi\include\sys\signal.h:42:7: error: redefinition of 'union sigval'
\env\tools\gnu_gcc\arm_gcc\mingw\arm-none-eabi\include\sys\signal.h:47:8: error: redefinition of 'struct sigevent'

BernardXiong, 2019-10-15:
主要是newlib这块的变动，目前需要用低版本的工具链，GCC6.x吧

FlyGoat, 2019-12-05:
@BernardXiong I just took a look on this issue. We have two solutions. The first is to use pre-defined compiler macros to overlay stdlib headers, But we have any to control the including order carefully. The second is to build a toolchain specified for RT-Thread without built-in newlib. It would be nice if we can do that by crosstool-ng and Travis-CI.

FlyGoat, 2019-12-05:
Or probably we can make our built-in minilibc more powerful so we don't have to link against newlib?
Grab code from musl would be a option.

BernardXiong, 2019-12-05:
Yes, we will consider maintaining a libc library. It's possible from musl libc.

FlyGoat, 2019-12-05:
> Yes, we will consider maintaining a libc library. It's possible from musl libc.

As far as I known. Newlib is licensed under GPL. Musl is licensed under MIT, which is possible to convert to Apache 2.
Do we have any other option that doesn't have legal issue?

BernardXiong, 2019-12-05:
It's better to use musl libc. We have an internal version of musl, but it's slow to build on Windows.

closed 2022-01-20 by mysterywolf