RT-Thread/rt-thread #3141: drv_pwm.c and drv_hwtimer.c build error: use of empty

LUAN-Z, 2019-10-19:
..\libraries\HAL_Drivers\drv_hwtimer.c(73): error:  #40: expected an identifier
  };
..\libraries\HAL_Drivers\drv_hwtimer.c(152): error:  #29: expected an expression
  };
..\libraries\HAL_Drivers\drv_hwtimer.c(83): error:  #1514: an empty initializer is invalid for an array with unspecified bound
  static struct stm32_hwtimer stm32_hwtimer_obj[] =
..\libraries\HAL_Drivers\drv_hwtimer.c: 0 warnings, 3 errors
compiling drv_gpio.c...
compiling drv_pwm.c...
..\libraries\HAL_Drivers\drv_pwm.c(78): error:  #40: expected an identifier
  };
..\libraries\HAL_Drivers\drv_pwm.c(157): error:  #29: expected an expression
  };
..\libraries\HAL_Drivers\drv_pwm.c(88): error:  #1514: an empty initializer is invalid for an array with unspecified bound
  static struct stm32_pwm stm32_pwm_obj[] =
..\libraries\HAL_Drivers\drv_pwm.c: 0 warnings, 3 errors

BernardXiong, 2019-10-21:
Please provide more information. Which case will cause this issue?

xuzheng821, 2021-02-18:
I found the same problem，when develop base the BSP of stm32f413-st-nucleo，is there any update?

xuzheng821, 2021-02-18:
> Please provide more information. Which case will cause this issue?

I found the same problem，when develop base the BSP of stm32f413-st-nucleo，is there any update? thanks!

mysterywolf, 2022-06-16:
[image]
你要在这里去打开对应的宏，想用PWM几，什么宏都不打开，只开启了PWM，肯定会报错的

closed 2022-06-16 by mysterywolf