RT-Thread/rt-thread #3144: Will  Nuvoton_M497 bsp support the Keil 5.28 version?

Sam-Wang, 2019-10-21:
Hi :
 Will  Nuvoton_M487 bsp support above the Keil 5.28 version ?
BRS

BernardXiong, 2019-10-21:
You can try it. If you not, you can use `scons --target=mdk5` to re-generate MDK project.

Sam-Wang, 2019-10-21:
The M487 bsp was base keil 4.74, No template.uvprojx to re-generate !
.... \rt-thread\bsp\nuvoton_m487\SConstruct", line 27, in <module>
IOError: [Errno 2] No such file or directory: 'template.uvprojx':

Sam-Wang, 2019-10-21:
Rename the template.uvproj  to template.uvprojx .
scons --target=mdk5  will generate a project.uvprojx file. 

".\build\template.axf" - 0 Error(s), 2 Warning(s).

Thanks !

BernardXiong, 2019-10-23:
Yes. The template project file is just an empty project file. Therefore, you can use Keil MDK to genernate a new and empty project file with correct device settings.

closed 2019-10-28 by BernardXiong