RT-Thread/rt-thread #3173: 建议增加标准错误信息输出函数

a1012112796, 2019-11-02:
各位大佬好，最近想到了一个小麻烦，就是RTT中没有标准错误信息输出函数，在发生错误时只能输出一个数字，然后只能通过查代码或者文档去找到这个数字所代表的含义。故建议增加strerror函数以增加错误信息输出的可读性。这是我尝试做的一个比较low的实现及输出效果，仅供参考。
一点个人意见，如有错误，还望各位大佬见谅，谢谢。
[add_strerror_patch.zip](https://github.com/RT-Thread/rt-thread/files/3800059/add_strerror_patch.zip)
[image]

BernardXiong, 2019-11-04:
谢谢。

请问有github repo可以看看是什么样实现的吗？另外，关于errno标准错误的部分，应该是可以用strerror函数的。

a1012112796, 2019-11-04:
@BernardXiong 
您好，我尝试做的改动在附件add_strerror_patch.zip中以patch补丁的形式给出，用的是最简单的数组的方式实现的并且复制了%s的很多内容，我觉得应该有更好的办法，所以没发，仅放在这供各位大什佬参考。

a1012112796, 2019-11-04:
[image]
还有请问现在是不是发ISSUE必须用英文了？！:cry:

BernardXiong, 2019-11-05:
> [image]
> 还有请问现在是不是发ISSUE必须用英文了？！😢

Yes, If you use Chinese, suggest to submit in [Chinese Forum](https://www.rt-thread.org/qa/portal.php) or [Gitee](https://gitee.com/rtthread/rt-thread).

a1012112796, 2019-11-05:
OK,I see. I have change it.

closed 2019-11-08 by BernardXiong