RT-Thread/rt-thread #3174: 编译失败

jiejiezhang, 2019-11-03:
你好，我按照官方文档操作配置ununtu下的环境：　https://www.rt-thread.org/document/site/application-note/setup/qemu/ubuntu/an0005-qemu-ubuntu/

最后出现如下错误，帮忙看一下是什么原因引起的，谢谢
[image]

a1012112796, 2019-11-03:
@jiejiezhang 您好，建议您在论坛上发。

jiejiezhang, 2019-11-03:
好的，非常感谢这么快的回复

BernardXiong, 2019-11-04:
It's caused by GCC version or GCC detection issue. 

Normally, there are two `fd_set` definitions, libc's or RT-Thread's. The scons will check which definition should be used and then generate cconfig.h file for compiling.

When you met this issue, you can:

1. Remove cconfig.h file manually or by `scons -c` command;

or 

2. Use old GNU GCC version (<= 6.x in ARM).

jiejiezhang, 2019-11-04:
Thanks very much. Now it can work normally!

closed 2019-11-05 by jiejiezhang