RT-Thread/rt-thread #3183: ctime()函数调用出错

essemi-yuzr, 2019-11-07:
程序运行到以下函数(time.c文件中)会进hardfault
`char* asctime_r(const struct tm *t, char *buf)
{
    /* "Wed Jun 30 21:49:08 1993\n" */
    *(int*) buf = *(int*) (days + (t->tm_wday << 2));
    *(int*) (buf + 4) = *(int*) (months + (t->tm_mon << 2));

    ......
}`

BernardXiong, 2019-11-08:
恶意单步跟踪下，具体是哪条语句，以及为什么会跑飞掉

essemi-yuzr, 2019-11-11:
是由于前面的
`char* asctime(const struct tm *timeptr)
{
    static char buf[25];
    return asctime_r(timeptr, buf);
}`函数在定义了一个25字节的数组，但是没有4字节对齐，所以后面到了asctime_r()函数中进行操作（函数中“(int) buf = (int) (days + (t->tm_wday << 2));”语句是按照4字节进行操作的）会有概率产生问题

essemi-yuzr, 2019-11-11:
是因为“time.c”文件中‘static const char days[]’数组是1字节对齐（我编译出来的地址是0x000073EC），所以按照'(int) buf = (int) (days + (t->tm_wday << 2));'4字节操作会有概率出错

essemi-yuzr, 2019-11-11:
是因为“time.c”文件中‘static const char days[]’以及'static const char months[]'数组是1字节对齐，在asctime_r()函数中是按照4字节操作的，这样存在一定的不对齐操作的风险，在我的程序中恰恰months数组在编译的时候不是4字节对齐（地址为0x000073A5），所以出错了，也就是在以上我列举的两句代码操作时出错的。

mysterywolf, 2021-02-05:
#3938

mysterywolf, 2021-02-09:
我将会处理这个问题，谢谢反馈。

closed 2021-02-11 by mysterywolf