RT-Thread/rt-thread #3215: Package for testing interrupt latencies!

adeldennaoui, 2019-11-21:
Hello!

I'd like to ask if there is any package that is similar to cyclictest for Linux that I can use for RT-Thread or port to be able to measure the interrupt latencies.. 
I am working on comparing the latencies of RTOS and RT-Thread should have a great potential so nay help would be appreciated..
I am running RT-Thread using QEMU to emulate qemu-vexpress-a9 and thank you! :)

BernardXiong, 2019-11-21:
It's hard to test interrupt latency. Normally, You can use an oscilloscope to measure the interrupt latency, for example to measure the time of two GPIO level. And,,, It's so complicated.

There is a simple method to measure the interrupt latency:

You can select a hardware timer. When this timer is timeout, it can tigger an interrupt. Then you can read the timer count in this interrupt service routine. This timer count is the interrupt latency.

adeldennaoui, 2019-11-25:
Can you please write me the code for the simple method? I'm relatively new to RT-Thread.. I'm learning the language but it would be appreciated to take a look of the code and try it right away. Thank you!

closed 2021-05-16 by mysterywolf