RT-Thread/rt-thread #3216: qemu-vexpress-a9 vs qemu-vexpress-gemini

adeldennaoui, 2019-11-21:
What's the difference when running RT-Thread ?

BernardXiong, 2019-11-21:
qemu-vexpress-a9: SMP cortex-a9 with SMP RT-Thread;
qemu-vexpress-gmini: dual cortex-a9 with linux + RT-Thread. They use vbus for operating system communication.

adeldennaoui, 2019-11-21:
And since there's a VBUS between Linux and RT-Thread so I assume that I may be able to use cyclictest to measure the interrupt latency of RT-Thread or am I being too cringy with those noob questions?

adeldennaoui, 2019-11-21:
Also I'd like to ask one thing, I applied "sudo scons" to qemu-vexpress-gemini and then ./qemu.sh, I can see the QEMU but I'm not getting how to switch to RT-Thread console..
In the case of realview-a8, i could've go to a linux console and login with root and then insert the module rtvmm.ko and then I get the ability to switch to RT-Thread but here, I don't get the linux console

BernardXiong, 2019-11-22:
Did you build Linux?

adeldennaoui, 2019-11-25:
No I didn't build it

closed 2021-05-16 by mysterywolf