RT-Thread/rt-thread #3219: 求助 IAR 下不能有重名文件的问题是怎样解决的呢？

SuWeipeng, 2019-11-25:
问题
---
我没有用 env ，在 IoT Board（潘多拉开发板）上搬出文件系统代码，IAR 下发现 time.h 重名。

问题代码和我的解决办法
---
> 路径：Libraries\rt-thread\components\libc\compilers\common\sys 

将 time.h 改为 rt_time.h

问题代码：
[pandora_filesystem.zip](https://github.com/RT-Thread/rt-thread/files/3884851/pandora_filesystem.zip)

问题重现：
1. 将 rt_time.h 改回 time.h 
2. \Libraries\rt-thread\components\libc\compilers\common\time.c 第 11 行改回  #include <sys/time.h>

编译不过的问题 100% 复现

我的疑惑
---
在 env 生成的 iar 工程里同样是 time.h 重名，但就是没有问题，怎么搞定的呢？

[image]

SuWeipeng, 2019-11-25:
[image]
工程在 generate_project 文件夹里，为了减小体积我生成工程后没编译，编译的话要按上图设置一下。

BernardXiong, 2019-11-25:
这个，，，IAR 重名的问题无解，貌似

armink, 2019-11-25:
我们一般是这样用的

`#include <sys/time.h>`

SuWeipeng, 2019-11-25:
> 这个，，，IAR 重名的问题无解，貌似

不会啊，用 env 生成的工程就没有问题，即使重名了也不影响。我单拿出来就不行。
# env 怎么这么神奇
# env 为什么这么神奇
# env 咋解决这个问题的呢

BernardXiong, 2019-11-25:
是不是路径顺序的问题，也就是以哪个优先次序来搜索

closed 2020-01-02 by BernardXiong