RT-Thread/rt-thread #3228: 串口接收有点乱码，波特率越高，乱码越多

Aric26, 2019-12-02:
该问题只在4.0的版本上出现，同样的配置3.0版本未出现该问题，使用的芯片是：STM32F030c8，时钟源选择HSI通过PLL倍频到48M，之后的选择基本都是0分频的，串口选择PCLK为时钟来源，PCLK为48M,寄存器如下：
RCC寄存器：CR 0x03005003 CFGR 0x002800A CFCR3 0x0   这里只是说明UART1的时钟频率和时钟源为：48M PCLK

UART1寄存器：CR1:0x2D BRR:0x1A1 ISR: 0X00600D0   这里说明：串口配置 数据8位 1位停止位 RX TX UE使能 波特率位115200

引脚配置：PB6 PB7 , GPIOB端口：MODER寄存器：0xA000  OTYPER: 0x0 OSPEEDR:0XF000  PUPDR:0X04005000  AFRL:0x0  使用PB6和PB7IO口 复用功能 速率最大 内部上拉

比如：NVIC  UART1中断使能

现象如下：
hello world
hello world
hello world
hello world
hello world
hello world
hello world
hello world
hello world
hello world
鑕llo world
鑕llo world
hellwor靌
hel靜 world
hel靜 world
hel靜 鱫rl
hello w飏ld
hello world
hello world
hello world
hello w飏ld
hello world
h錶lo world
hello world

Aric26, 2019-12-02:
排除硬件板子问题，同样的板子，刷入3.0的版本（串口的配置一模一样）并未此问题

BernardXiong, 2019-12-02:
Please provide more clear information about this issue.

Aric26, 2019-12-02:
What kind of information do you need?

> Please provide more clear information about this issue.

What kind of information do you need?

balanceTWK, 2019-12-02:
115200 波特率会有问题？

@Aric26  

https://github.com/RT-Thread/rt-thread/blob/master/bsp/stm32/libraries/HAL_Drivers/drv_usart.c
和这个链接里的驱动对比下，是最新的驱动嘛？

Aric26, 2019-12-03:
> 115200 波特率会有问题？
> 
> @Aric26
> 
> https://github.com/RT-Thread/rt-thread/blob/master/bsp/stm32/libraries/HAL_Drivers/drv_usart.c
> 和这个链接里的驱动对比下，是最新的驱动嘛？

我对比了一下，跟你的链接的驱动不一样，但是替换你这个驱动，问题仍然存在

BernardXiong, 2019-12-05:
Please provide more information, such as:

1. Which baudrate? 
2. version information.
3. toolchains information.
4. test code.

Aric26, 2019-12-06:
1、the baud rate is 115200
2、RT-Thread v4.0.1
3、ARM RealView(armcc)
4、rt_kprintf("hello world\n");

closed 2022-08-14 by mysterywolf