RT-Thread/rt-thread #3229: stm32下的编码器设备问题！

sogwms, 2019-12-02:
在实际使用时发现系统随着编码器所测电机转速的逐渐增加竟越来越卡顿，最终排查是编码器设备的问题，粗略看了下，在编码器的底层 driver 中竟然启动了引脚翻转中断， 要知道其引脚翻转速率是跟随电机转速， 一般速度的电机都能 “卡死” 系统。 望修复

（最后想说一点， 要实现编码器设备必然要依赖“编码器硬件”， 单纯靠定时器输入捕获等中断方式软计数在实际使用是不可取的）

balanceTWK, 2019-12-02:
https://github.com/RT-Thread/rt-thread/blob/master/bsp/stm32/libraries/HAL_Drivers/drv_pulse_encoder.c

额，官方的stm32编码器驱动没有使用引脚翻转中断来计数，而是使用了 stm32定时器的 AB 相编码器模式来计数。

（驱动里只开了定时器溢出中断，作用是 计数超过 +-32767 后，仍能累加计数。 ）

sogwms, 2019-12-02:
仔细看了下，是的，确实是用了定时器的编码器模式， 但在使用中确实有问题， 现在直接使用HAL库没有出现上述卡顿情况

balanceTWK, 2019-12-02:
[rt_err_t pulse_encoder_init(struct rt_pulse_encoder_device *pulse_encoder) ](https://github.com/RT-Thread/rt-thread/blob/master/bsp/stm32/libraries/HAL_Drivers/drv_pulse_encoder.c#L80)

对比下初始化函数的区别吧，rtt 用的也是 hal 库。

sogwms, 2019-12-02:
我知道， 看着都是没什么问题的。 可使用不一样啊

sogwms, 2019-12-02:
还有 从 0逆向到重装值 与 正向都 会引发更新事件吧， 此时 driver 的计数是否不妥

balanceTWK, 2019-12-02:
> 还有 从 0逆向到重装值 与 正向都 会引发更新事件吧， 此时 driver 的计数是否不妥

不会，中断里有判断是向上溢出还是向下溢出的。https://github.com/RT-Thread/rt-thread/blob/master/bsp/stm32/libraries/HAL_Drivers/drv_pulse_encoder.c#L181

sogwms, 2019-12-02:
此处呢 https://github.com/RT-Thread/rt-thread/blob/0dc7da688fa4d20745461b8835d3f6228fb90bf3/bsp/stm32/libraries/HAL_Drivers/drv_pulse_encoder.c#L146

sogwms, 2019-12-02:
结合这个地方
https://github.com/RT-Thread/rt-thread/blob/0dc7da688fa4d20745461b8835d3f6228fb90bf3/bsp/stm32/libraries/HAL_Drivers/drv_pulse_encoder.c#L137

sogwms, 2019-12-02:
结合这里
https://github.com/RT-Thread/rt-thread/blob/0dc7da688fa4d20745461b8835d3f6228fb90bf3/bsp/stm32/libraries/HAL_Drivers/drv_pulse_encoder.c#L138

balanceTWK, 2019-12-02:
> 结合这里
> https://github.com/RT-Thread/rt-thread/blob/0dc7da688fa4d20745461b8835d3f6228fb90bf3/bsp/stm32/libraries/HAL_Drivers/drv_pulse_encoder.c#L138

没问题啊

sogwms, 2019-12-02:
额，    我今天怕是晕了，  对的， 没问题。  虽然还是“卡”

closed 2019-12-02 by sogwms

balanceTWK, 2019-12-02:
> 额， 我今天怕是晕了， 对的， 没问题。 虽然还是“卡”

https://github.com/RT-Thread/rt-thread/blob/0dc7da688fa4d20745461b8835d3f6228fb90bf3/bsp/stm32/libraries/HAL_Drivers/drv_pulse_encoder.c#L112

可以试下把驱动里的溢出中断也关了。