RT-Thread/rt-thread #3239: MF1 AI + IoT board problem

TheArheus, 2019-12-05:
Hello.

I have a problem with my mf1 AI+IoT board. I accidentally tried to
flash .elf file to board and after realizing that this is wrong file,
I cancelled flash. After that my board is not working properly.

I can flash any .bin files now, but after flashinh programms with lcd
output, the only collor on lcd is red. On another boards camera and
lcd works fine, but It seems, that they're not working on this board
now.

This board just in infinite loop and in serial port outputs this:

```
manuf_id:0xc8, device_id:0x17
The header ok
there is 0 img
wdt reboot!firmware version:
0.0.0
face_lib_version:
2019-11-14 11:14:57
FAST-4-gcce67666a307cd0e-dirty
dump board cfg:
    out_threshold    : 88.000000
    auto_out_fea     : 0
    out_fea          : 0
    out_interval_ms  : 100
    pkt_fix          : 0
    port_baud        : 115200
    relay_open_s     : 2
    cam_flip         : 1
    cam_hmirror      : 0
    lcd_dir          : 0
    lcd_flip         : 0
    lcd_hmirror      : 0
    key              : 24
    key_dir          : 1
    log_rx           : 11
    log_tx           : 10
    port_rx          : 4
    port_tx          : 5
    relay_high       : 12
    relay_low        : 13
board key relay cfg:
	key:24	key_dir:1	relay_high:13	relay_low:12
board cfg lcd cam:
	lcd_flip:0,lcd_hmirror:0,cam_flip:1,cam_hmirror:0
lcd_set_direction: 00
camera_set_hmirror: 0
camera_set_vflip: 1
board uart pin cfg:
	port_baud:115200,port_tx:5,port_rx:4,log_tx:10,log_rx:11
{"version":2,"type":"init","code":0,"msg":"init done","proto":"0.1"}
```

How can I fix this problem?

BernardXiong, 2019-12-05:
mf1 AI+IoT board, What's that?

TheArheus, 2019-12-06:
It is this board: https://www.aliexpress.com/i/4000070597568.html

BernardXiong, 2019-12-09:
Unknow board. Is it runing RT-Thread? For the board issue, you can ask board manufactory.

closed 2019-12-09 by BernardXiong