RT-Thread/rt-thread #3248: lwip的pbuf_free有点问题

Simon-Leung, 2019-12-08:
pbuf.c里面的pbuf_free函数，650行里的LWIP_ASSERT("pbuf_free: p->ref > 0", p->ref > 0)，建立UDP服务端后，以小于1ms间隔不断给RTT系统发送数据，结果就卡在这个assert里了。

BernardXiong, 2019-12-09:
具体是什么问题？

Simon-Leung, 2019-12-11:
具体应用场景是，RTT终端通过以太网口作为一个NTP服务器。
PC机打开TCP调试助手，向NTP服务器发送NTP请求授时的UDP包，然后NTP服务器向TCP调试助手发送授时的UDP包。
当TCP助手发送频率加大到ms级，NTP服务器就会死机，调试发现卡在`LWIP_ASSERT("pbuf_free: p->ref > 0", p->ref > 0)`，p->ref的值为0。
而且，并非发送频率到ms级才会出现，1秒钟的频率运行一段时间也会出现上述现象，就是说，频率越高，死机越快。

Simon-Leung, 2019-12-11:
后来我把这句断言'LWIP_ASSERT("pbuf_free: p->ref > 0", p->ref > 0)'注释掉了才能稳定运行

BernardXiong, 2019-12-12:
是否有能够100%复现的代码或方法？

Simon-Leung, 2019-12-13:
你们测试都没问题？我这是100%复现的，我用的tm4c129x的BSP，会不会跟这个有关？

Simon-Leung, 2019-12-13:
我贴一部分代码上来

`fd = socket(AF_INET, SOCK_DGRAM, 0);`

```
void request_process_loop(int fd)
{
    struct sockaddr src_addr;
    socklen_t src_addrlen = sizeof(src_addr);
    uint8_t buf[48];
    uint32_t recv_time[2];
    struct sockaddr_in sinaddr;
    struct timeval tm;

    tm.tv_sec = 1;
    tm.tv_usec = 0;
    setsockopt(fd, SOL_SOCKET, SO_RCVTIMEO, &tm, sizeof(tm));

    memset(&sinaddr, 0, sizeof(sinaddr));
    sinaddr.sin_family = AF_INET;
    sinaddr.sin_port = htons(123);
    sinaddr.sin_addr.s_addr = htonl (INADDR_ANY);

    if (0 != bind(fd, (struct sockaddr *)&sinaddr, sizeof(sinaddr)))
    {
        LOG_E("Bind error");
        closesocket(fd);
        return;
    }

    while (1)
    {
        while (recvfrom(fd, buf,
                48, 0,
                &src_addr,
                &src_addrlen)
            < 48 )  /* invalid request */
        {
            if(param_get_ntpmode() != 0) /* 与TCP协议栈无关 */
            {
                return;
            }
        }
        sendto( fd, buf, 48, 0, &src_addr , src_addrlen);
    }
}
```

tamerxu, 2021-10-16:
https://github.com/RT-Thread/rt-thread/issues/568   此问题与#568类似，可以归集讨论分析

closed 2022-01-28 by mysterywolf