RT-Thread/rt-thread #3249: serial: Polling mode with shell is broken

FlyGoat, 2019-12-08:
I implemented a UART driver on an 8250 like device without interrupt support. However I found that I can't input anything to finsh. Further debug revaled that 'uart_getc' in this driver was never called, because callbacks relied on interrupt.

Any driver without RT_DEVICE_FLAG_INT_RX flag can probably reproduce the issue.

BernardXiong, 2019-12-09:
No driver like that for shell. If the RX interrupt broken, you can try to monitor rx data in idle thread and release the corresponding semaphore.

FlyGoat, 2019-12-09:
> No driver like that for shell. If the RX interrupt broken, you can try to monitor rx data in idle thread and release the corresponding semaphore.

Thanks for your suggestion. But I think this feature should be a part of components/serial infrastructure. Otherwise 'polling mode' shouldn't exist as it can never really work.

chenguokai, 2020-06-16:
I found a workaround for this.
Since we have polling mode available, adding the generic serial handler in the timer Interrupt works good for me.
The drawback is some unnecessary memory access.

FlyGoat, 2020-06-17:
> 
> 
> I found a workaround for this.
> Since we have polling mode available, adding the generic serial handler in the timer Interrupt works good for me.
> The drawback is some unnecessary memory access.

I'd prefer in idle thread, timer interrupts come in a very high frequency. That's too much overhead.

chenguokai, 2020-06-17:
> That's too much overhead.
That’s indeed a problem but I would like to provide 2 reasons for this.

1. Polling mode only serials seem to be restricted to limited number of boards. Exposing device specific operations to a more or less generic part of the system sounds terrible.
2. By implementing a naive counter in the interrupt handling process (in bsp or libcpu) and invoking the polling functions only when the counter hit a given point decreases the overhead and is more predictable.

closed 2022-06-13 by mysterywolf