RT-Thread/rt-thread #3259: stm32 can2的相关问题

2084046485, 2019-12-11:
最近做东西用到了stm32f4的can，发现can2无法接收，看寄存器配置发现是can2的筛选器相关的寄存器没使能。
最后发现在drv_can.c里对can的筛选器配置时调用了HAL_CAN_ConfigFilter函数
但在传入过滤器相关参数时默认FilterBank =0；而没有使能can2 对应的过滤器组。
原代码如下：
`
static rt_err_t drv_configure(struct rt_can_device *dev_can,
                              struct can_configure *cfg)
{
    rt_uint32_t baud_index;
	struct stm32_drv_can *drv_can;
    CAN_InitTypeDef *drv_init;
    CAN_FilterTypeDef *filterConf;
    RT_ASSERT(dev_can);
    RT_ASSERT(cfg);

    drv_can = (struct stm32_drv_can *)dev_can->parent.user_data;
    drv_init = &drv_can->CanHandle.Init;

    drv_init->TimeTriggeredMode = DISABLE;
    drv_init->AutoBusOff = ENABLE;
    drv_init->AutoWakeUp = DISABLE;
    drv_init->AutoRetransmission = ENABLE;
    drv_init->ReceiveFifoLocked = DISABLE;
    drv_init->TransmitFifoPriority = DISABLE;

    switch (cfg->mode)
    {
    case RT_CAN_MODE_NORMAL:
        drv_init->Mode = CAN_MODE_NORMAL;
        break;
    case RT_CAN_MODE_LISEN:
        drv_init->Mode = CAN_MODE_SILENT;
        break;
    case RT_CAN_MODE_LOOPBACK:
        drv_init->Mode = CAN_MODE_LOOPBACK;
        break;
    case RT_CAN_MODE_LOOPBACKANLISEN:
        drv_init->Mode = CAN_MODE_SILENT_LOOPBACK;
        break;
    }

    baud_index = get_can_baud_index(cfg->baud_rate);
    drv_init->SyncJumpWidth = BAUD_DATA(SJW, baud_index);
    drv_init->TimeSeg1 = BAUD_DATA(BS1, baud_index);
    drv_init->TimeSeg2 = BAUD_DATA(BS2, baud_index);
    drv_init->Prescaler = BAUD_DATA(RRESCL, baud_index);
    if (HAL_CAN_Init(&drv_can->CanHandle) != HAL_OK)
    {
        return RT_ERROR;
    }
    if (HAL_CAN_Start(&drv_can->CanHandle) != HAL_OK)
    {
        return RT_ERROR;
    }
    /* Filter conf */
	
    filterConf = &drv_can->FilterConfig;
    filterConf->FilterBank = 0;
    filterConf->FilterMode = CAN_FILTERMODE_IDMASK;
    filterConf->FilterScale = CAN_FILTERSCALE_32BIT;
    filterConf->FilterIdHigh = 0x0000;
    filterConf->FilterIdLow = 0x0000;
    filterConf->FilterMaskIdHigh = 0x0000;
    filterConf->FilterMaskIdLow = 0x0000;
    filterConf->FilterFIFOAssignment = CAN_FILTER_FIFO0;
    filterConf->FilterActivation = ENABLE;
    filterConf->SlaveStartFilterBank = 14;
    HAL_CAN_ConfigFilter(&drv_can->CanHandle, filterConf);
    return RT_EOK;}
`
修改之后，在该初始化函数中添加了以下代码

`
if(dev_can->parent.parent.name[3] == '2')
	{
		filterConf->FilterBank = 14;
		HAL_CAN_ConfigFilter(&drv_can->CanHandle, filterConf);
	}
`
这段代码的目的是识别当前初始化的设备是否为can2，如果是，通过HAL_CAN_ConfigFilter将can寄存器中FS1R，FFA1R,FA1R寄存器中can2筛选器对应的位置1，究竟是那位取决于传入的filterConf->FilterBank的值。

以上，can2无法接收的问题解决.（目前只在stm32f427上试验过）

BernardXiong, 2019-12-12:
谢谢反馈，欢迎发出PR出来 🥇

2084046485, 2019-12-15:
忘了说 这个问题出现在3.1.3的release版本中

2084046485, 2019-12-15:
并且这个版本里 由于f4的hal库在can的底层上相较于过去的版本有了较大改变，使用can的时候设备驱动层与hal库的底层并不对应。

closed 2022-06-08 by mysterywolf