RT-Thread/rt-thread #3272: 互斥锁嵌套使用中的优先级翻转问题?

13824125580, 2019-12-17:
基于lts分支测试如下代码：
#include <typedef.h>
#include <rtthread.h>
#include <waitqueue.h>
#include <rthw.h>
#include <arch.h>
#include <log.h>

void udelay(unsigned int us);
static rt_mutex_t mutex_s1, mutex_s2;
static void pri_task_a(void *ARG_UNUSED(para))
{
    while (1)
    {
        rt_thread_delay(2000);
        __log("before got s1");
        rt_mutex_take(mutex_s1, RT_WAITING_FOREVER);
        __log("got s1");
        udelay(100 * 1000 * 1000);
        __log("after delay 100s");
        rt_mutex_release(mutex_s1);
        __log("release s1");
    }
}

static void pri_task_b(void *ARG_UNUSED(para))
{
    rt_thread_delay(5000);

    while (1)
    {
	udelay(1);
    }
}

static void pri_task_c(void *ARG_UNUSED(para))
{
    while (1)
    {
        rt_thread_delay(1000);
        __log("before got s1");
        rt_mutex_take(mutex_s1, RT_WAITING_FOREVER);
        __log("got s1");
        rt_mutex_take(mutex_s2, RT_WAITING_FOREVER);
        __log("got s2");
        udelay(100 * 1000 * 1000);
        __log("delay 100s");
        rt_mutex_release(mutex_s2);
        __log("release s2");
        rt_mutex_release(mutex_s1);
        __log("release s1");
    }
}

static void pri_task_d(void *ARG_UNUSED(para))
{
    while (1)
    {
        __log("before got s2");
        rt_mutex_take(mutex_s2, RT_WAITING_FOREVER);
        __log("got s2");
        udelay(100 * 1000 * 1000);
        __log("after delay 100s.");
        rt_mutex_release(mutex_s2);
        __log("release s2.");
    }
}

/* ----------------------------------------------------------------------------*/
/*
 *  @brief  schedule_in_irqlock <test pattern, Thread swith out when iterrupt
 *          disable>
 */
/* ----------------------------------------------------------------------------*/
void priority_entry(void)
{
    rt_thread_t thread;

    mutex_s1 = rt_mutex_create("mutexs1", RT_IPC_FLAG_FIFO);
    mutex_s2 = rt_mutex_create("mutexs2", RT_IPC_FLAG_FIFO);
    if(mutex_s1 == RT_NULL || mutex_s2 == RT_NULL)
    {
        __err("fatal error");
        return;
    }

    thread = rt_thread_create("pri_a", pri_task_a, RT_NULL, 0x1000, 0, 10);
    rt_thread_startup(thread);

    thread = rt_thread_create("pri_b", pri_task_b, RT_NULL, 0x1000, 1, 10);
    rt_thread_startup(thread);

    thread = rt_thread_create("pri_c", pri_task_c, RT_NULL, 0x1000, 2, 10);
    rt_thread_startup(thread);

    thread = rt_thread_create("pri_d", pri_task_d, RT_NULL, 0x1000, 3, 10);
    rt_thread_startup(thread);

    return;
}
－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－
现象：
[DBG]: [pri_task_d:0075]: before got s2
[DBG]: [pri_task_d:0077]: got s2
[DBG]: [pri_task_c:0057]: before got s1
[DBG]: [pri_task_c:0059]: got s1
[DBG]: [pri_task_a:0032]: before got s1
－－－－－－－－－－－－－－－－－－－－－－－－
系统输出最后一行后，打印停止，finsh不在交互，处于永久锁死状态

测试条件：
A,B,C,D四个任务，mutexs1, mutexs2两个信号量.
任务优先级A>B>C>D
程序流程：
1.D 执行，获取S2
2.C被唤醒，抢占D
3. C获取S1成功，但2已经被D占有，Ｃ被阻塞，D优先级继承C.
4. D继续执行，但中途Ａ被唤醒，Ａ抢占Ｄ
5. A尝试占有S1, 但Ｓ1此时被C占有，A阻塞，C优先级继承了Ａ，但扔阻塞在S2上
6.Ｄ继续执行，中途Ｂ被唤醒，Ｂ抢占了Ｄ
7. 由于Ｂ不需要占用任何资源，所以高优先级任务Ａ被中间优先级任务Ｂ阻塞，优先级继承失效。

－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－
我可以理解可以通过让B睡眠一段时间的方式解锁这个过程，但机制上，A的唤醒时间依赖于B何时去主动出让处理器，而这个条件完全由Ｂ决定，Ａ虽然最高优先级但无法保证deadline。高优先级任务Ａ的优先级没有传染给依赖链。

GFWisshit, 2019-12-18:
你说的对，他这个mutex实现的太简单了。完善的方法 应该是当线程无法获得一个互斥量的时候应该去检查这个互斥量的owner是否因为等待其他互斥量而阻塞，如果是的话就需要递归的提升互斥量owner的优先级。怎么改进，方法有很多，核心就是能判断owner是否也阻塞在其他互斥量上并且能找到那个互斥量的owner就行了。

13824125580, 2019-12-18:
@GFWisshit 可否提供一个规避此类问题的实现案例？

GFWisshit, 2019-12-18:
> @GFWisshit 可否提供一个规避此类问题的实现案例？

最简单的手段就是做操作系统的自己去改mutex实现，或者他再提供一个能同时lock多个mutex的api。
你这种情况，应用层能做到的就是你让A D也按照C的顺序去拿锁 不管用不用得到，或者优化代码逻辑，拿一个锁把相应的事干完了，释放掉再拿另一个锁，自己人肉保证不会同时lock两个mutex。

13824125580, 2019-12-18:
@GFWisshit 应用需求好说，这里提是想和大家交流学习RTT.
看了下代码，ucos2和ucos3都没有这种机制，可能发现的这种case太小众了，大家都不觉得的是太大问题。但是从硬实时性角度，能够改进当然最好。

我的理解：当前的运行任务要么是具有最高原始优先级的任务，要么是一个占有资源的低原始优先级任务，后者占用了更高优先级等待的资源。按照这个标准，上面的确实是一个问题。

13824125580, 2019-12-18:
＠GFWisshit 可以留个联系方式吗？一起交流学习RTT

GFWisshit, 2019-12-19:
> @GFWisshit 应用需求好说，这里提是想和大家交流学习RTT.
> 看了下代码，ucos2和ucos3都没有这种机制，可能发现的这种case太小众了，大家都不觉得的是太大问题。但是从硬实时性角度，能够改进当然最好。
> 
> 我的理解：当前的运行任务要么是具有最高原始优先级的任务，要么是一个占有资源的低原始优先级任务，后者占用了更高优先级等待的资源。按照这个标准，上面的应该是一个问题。

https://pubs.opengroup.org/onlinepubs/007908799/xsh/pthread_mutexattr_setprotocol.html
When a thread makes a call to pthread_mutex_lock(), the mutex was initialized with the protocol attribute having the value PTHREAD_PRIO_INHERIT, when the calling thread is blocked because the mutex is owned by another thread, that owner thread shall inherit the priority level of the calling thread as long as it continues to own the mutex. The implementation shall update its execution priority to the maximum of its assigned priority and all its inherited priorities. Furthermore, if this owner thread itself becomes blocked on another mutex, the same priority inheritance effect shall be propagated to this other owner thread, in a recursive manner.
我对这些东西都懒的吐槽了，自己明白咋回事就行了。这种实现，封装成posix，都满足不了posix的要求。

snikeguo, 2019-12-23:
@GFWisshit  freertos中有无这种的规避代码？或者其他的解决方案 来解决上述优先级翻转的问题

GFWisshit, 2019-12-23:
> @GFWisshit freertos中有无这种的规避代码？或者其他的解决方案 来解决上述优先级翻转的问题

没有。这种问题，没法简单的修改代码就能解决，即使能解决死锁，代价也是不可接受的，我说的那个本质也是错的，你可以去看相关的论文。

13824125580, 2019-12-24:
@GFWisshit "我说的那个本质也是错的"
感觉可行啊，哪里有问题吗？

closed 2019-12-26 by BernardXiong