RT-Thread/rt-thread #3285: W600 fail at wifi join , rtt 4.0.2

wendal, 2019-12-21:
```
msh />wifi join uiot AABBCCDDEEFF
[32m[18805] I/WLAN.mgnt: wifi connect success ssid:uiot

[0mmsh />
msh />[32m[19814] I/WLAN.lwip: Got IP address : 192.168.1.104

[0mthread:tcpip stack overflow
thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
task-05   14  suspend 0x000000c8 0x00000200    39%   0x00000014 000
task-04   13  suspend 0x000000a8 0x00002000    14%   0x00000008 000
task-03   10  suspend 0x000000a0 0x00000200    31%   0x00000014 000
task-02    7  suspend 0x000001dc 0x00000640    53%   0x00000012 000
task-01    4  suspend 0x00000174 0x000004b0    52%   0x0000000e 000
task-00    5  suspend 0x000000b4 0x00000320    56%   0x00000001 000
tshell    20  suspend 0x00000180 0x00001000    25%   0x00000009 000
ntp_sync  26  suspend 0x0000009c 0x00000600    10%   0x00000002 000
sys_work  23  suspend 0x000001c4 0x00000800    26%   0x0000000a 000
hostspi   19  suspend 0x000000d4 0x00000200    42%   0x00000013 000
wlan_job  22  suspend 0x00000080 0x00000800    43%   0x00000003 000
tcpip     10  running 0x000000e4 0x00000400    100%   0x0000000e 000
tidle0    31  ready   0x00000058 0x00000100    51%   0x0000001b 000
timer      4  suspend 0x000000a4 0x00000200    51%   0x00000005 000
```

so, I change `RT_LWIP_TCPTHREAD_STACKSIZE` in rtconfig.h , it works

```
[0mmsh />wifi join uiot AABBCCDDEEFF
[32m[7687] I/WLAN.mgnt: wifi connect success ssid:uiot

[0mmsh />
msh />[32m[8696] I/WLAN.lwip: Got IP address : 192.168.1.104

[0mlist_thread
thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
task-05   14  suspend 0x000000c8 0x00000200    39%   0x00000014 000
task-04   13  suspend 0x000000a8 0x00002000    14%   0x00000006 000
task-03   10  suspend 0x000000a0 0x00000200    31%   0x00000014 000
task-02    7  suspend 0x000001cc 0x00000640    47%   0x00000007 000
task-01    4  suspend 0x00000174 0x000004b0    52%   0x00000007 000
task-00    5  suspend 0x000000b4 0x00000320    56%   0x00000002 000
tshell    20  running 0x00000250 0x00001000    25%   0x00000001 000
ntp_sync  26  suspend 0x0000009c 0x00000600    10%   0x00000002 000
sys_work  23  suspend 0x00000084 0x00000800    52%   0x00000009 000
hostspi   19  suspend 0x000000d4 0x00000200    42%   0x00000013 000
wlan_job  22  suspend 0x00000080 0x00000800    42%   0x00000002 000
tcpip     10  suspend 0x000000e4 0x00001000    25%   0x0000000b 000
tidle0    31  ready   0x00000058 0x00000100    51%   0x0000001f 000
timer      4  suspend 0x000000a4 0x00000200    51%   0x00000008 000
```

armink, 2019-12-23:
堆栈发生移除了，可以增大 tcpip 堆栈试试

另外建议 W600多 使用 SDK ，https://github.com/RT-Thread/W601_IoT_Board

wendal, 2019-12-24:
我想表达的是, 默认配置就挂, 这就不好了

closed 2021-05-15 by wendal