RT-Thread/rt-thread #3286: 关于netdev中IEMI的问题

zcj20080882, 2019-12-22:
看代码发现netdev处理`hwaddr`时，有两种类型，分别是“MAC”和“IEMI”。确定是“IEMI”而不是“IMEI”？是故意这样处理的吗？
```c
/* 6 - MAC address, 8 - IEMI */
        if (netdev->hwaddr_len == NETDEV_IFCONFIG_MAC_MAX_LEN)
        {
            rt_kprintf("MAC: ");
            for (index = 0; index < netdev->hwaddr_len; index++)
            {
                rt_kprintf("%02x ", netdev->hwaddr[index]);
            }
        }
        else if (netdev->hwaddr_len == NETDEV_IFCONFIG_IEMI_MAX_LEN)
        {
            rt_kprintf("IEMI: ");
            for (index = 0; index < netdev->hwaddr_len; index++)
            {
                /* two numbers are displayed at one time*/
                if (netdev->hwaddr[index] < 10 && index != netdev->hwaddr_len - 1)
                    rt_kprintf("0");
                
                rt_kprintf("%d", netdev->hwaddr[index]);
            }
        }
```

Lawlieta, 2019-12-23:
这里之前是笔误，现在最新分支已经改成 IMEI 了，如下链接所示：

https://github.com/RT-Thread/rt-thread/blob/master/components/net/netdev/src/netdev.c#L890-L901

zcj20080882, 2019-12-24:
嗯，我在最新发布的V4.0.2上看到已经修改了

closed 2019-12-24 by zcj20080882