RT-Thread/rt-thread #3300: lpc55s69_nxp_evk build fail

karl-zh, 2019-12-30:
Keil version: 5.28
Code base: commit 8d0ca47526d23e8191139b304b6050c5eacf7401
Project: rt-thread\bsp\lpc55sxx\lpc55s69_nxp_evk

**Issue 1**: including file error

board\board.h(22): error: 'clock_config.h' file not found
#include "clock_config.h"
         ^~~~~~~~~~~~~~~~

**Issue 2**: symbol missing
.\build\rtthread-lpc55s6x.axf: Error: L6218E: Undefined symbol i2c_dbg (referred from drv_i2c.o).

BernardXiong, 2020-01-02:
@Magicoe 

Hi Magicoe,
There is also no link script file for GCC building:

`cannot open linker script file board/linker_scripts/link.lds`

I fix them with #3306, please check it.

karl-zh, 2020-01-08:
> @Magicoe
> 
> Hi Magicoe,
> There is also no link script file for GCC building:
> 
> `cannot open linker script file board/linker_scripts/link.lds`
> 
> I fix them with #3306, please check it.

Hi @BernardXiong ,

After the #3306 merged, there are new issues pop for ARM CC as no definition of __HeapLimit.

`board/board.c(95): error: use of undeclared identifier '__HeapLimit'
    rt_kprintf("sram heap, begin: 0x%p, end: 0x%p\n", HEAP_BEGIN, HEAP_END);
                                                                  ^
board/board.h(43): note: expanded from macro 'HEAP_END'
#define HEAP_END    ((void*)&__HeapLimit)
                             ^
board/board.c(96): error: use of undeclared identifier '__HeapLimit'
    rt_system_heap_init((void *)HEAP_BEGIN, (void *)HEAP_END);
                                                    ^
board/board.h(43): note: expanded from macro 'HEAP_END'
#define HEAP_END    ((void*)&__HeapLimit)
                             ^
2 errors generated.
compiling board.c...
".\build\rtthread-lpc55s6x.axf" - 2 Error(s), 0 Warning(s).`

>

closed 2021-02-06 by mysterywolf