RT-Thread/rt-thread #3304: mem、slab算法实现问题

mttbx, 2020-01-01:
你好，使用rtthread发现以下几点问题，希望能获得帮助：
1. slab算法分配了两次内存就分配失败了，实际分配大小占可分配大小应该不足3%.
2. mem算法要好一些，分配到最后实际分配大小占可分配大小不到50%.
这两种算法感觉都有缺陷。

以下是mem的调试记录：
```
[18:43:56:863] ␍␊
[18:43:56:863]  \ | /␍␊
[18:43:56:863] - RT -     Thread Operating System␍␊
[18:43:56:870]  / | \     4.0.2 build Jan  1 2020␍␊
[18:43:56:870]  2006 - 2019 Copyright by rt-thread team␍␊
[18:43:56:875] malloc size 168␍␊
[18:43:56:878] allocate memory at 0x2000d7d8, size: 180␍␊
[18:43:57:955] big mem!, 2048, 180, 180, 75804␍␊
[18:43:57:958] malloc size 2048␍␊
[18:43:57:958] allocate memory at 0x2000d88c, size: 2060␍␊
[18:43:57:965] malloc size 68␍␊
[18:43:57:965] allocate memory at 0x2000e098, size: 80␍␊
[18:43:57:969] malloc size 32␍␊
[18:43:57:973] allocate memory at 0x2000e0e8, size: 44␍␊
[18:43:57:977] malloc size 56␍␊
[18:43:57:980] allocate memory at 0x2000e114, size: 68␍␊
[18:43:57:980] malloc size 168␍␊
[18:43:57:980] allocate memory at 0x2000e158, size: 180␍␊
[18:43:58:265] big mem!, 1024, 2612, 2612, 75804␍␊
[18:43:58:268] malloc size 1024␍␊
[18:43:58:268] allocate memory at 0x2000e20c, size: 1036␍␊
[18:43:58:278] lwIP-2.1.0 initialized!␍␊
[18:43:58:278] malloc size 5, but align to 8␍␊
[18:43:58:278] allocate memory at 0x2000e618, size: 24␍␊
[18:43:58:469] big mem!, 6096, 3672, 3672, 75804␍␊
[18:43:58:472] malloc size 6096␍␊
[18:43:58:472] allocate memory at 0x2000e630, size: 6108␍␊
[18:43:58:636] big mem!, 6096, 9780, 9780, 75804␍␊
[18:43:58:640] malloc size 6096␍␊
[18:43:58:640] allocate memory at 0x2000fe0c, size: 6108␍␊
[18:43:58:650] malloc size 128␍␊
[18:43:58:650] allocate memory at 0x200115e8, size: 140␍␊
[18:43:58:650] malloc size 128␍␊
[18:43:58:655] allocate memory at 0x20011674, size: 140␍␊
[18:43:58:660] malloc size 76␍␊
[18:43:58:660] allocate memory at 0x20011700, size: 88␍␊
[18:43:58:665] malloc size 92␍␊
[18:43:58:665] allocate memory at 0x20011758, size: 104␍␊
[18:43:58:721] malloc size 52␍␊
[18:43:58:721] allocate memory at 0x200117c0, size: 64␍␊
[18:43:58:727] malloc size 168␍␊
[18:43:58:734] allocate memory at 0x20011800, size: 180␍␊
[18:43:58:979] big mem!, 1024, 16604, 16604, 75804␍␊
[18:43:58:982] malloc size 1024␍␊
[18:43:58:982] allocate memory at 0x200118b4, size: 1036␍␊
[18:43:58:990] malloc size 132␍␊
[18:43:58:990] allocate memory at 0x20011cc0, size: 144␍␊
[18:43:58:995] malloc size 180␍␊
[18:43:58:995] allocate memory at 0x20011d50, size: 192␍␊
[18:43:58:999] malloc size 68␍␊
[18:43:59:003] allocate memory at 0x20011e10, size: 80␍␊
[18:43:59:006] malloc size 168␍␊
[18:43:59:010] allocate memory at 0x20011e60, size: 180␍␊
[18:43:59:203] big mem!, 2048, 18236, 18236, 75804␍␊
[18:43:59:206] malloc size 2048␍␊
[18:43:59:206] allocate memory at 0x20011f14, size: 2060␍␊
[18:43:59:213] malloc size 56␍␊
[18:43:59:213] allocate memory at 0x20012720, size: 68␍␊
[18:43:59:216] hit␍␊
[18:43:59:220] malloc size 16␍␊
[18:43:59:220] allocate memory at 0x20012764, size: 28␍␊
[18:43:59:223] hit␍␊
[18:43:59:223] <6>Jan  1 00:00:00 sal.skt: Socket Abstraction Layer initialize success.␍␍␊
[18:43:59:231] malloc size 428␍␊
[18:43:59:231] allocate memory at 0x20012780, size: 440␍␊
[18:43:59:237] hit␍␊
[18:43:59:237] malloc size 16␍␊
[18:43:59:237] allocate memory at 0x20012938, size: 28␍␊
[18:43:59:243] hit␍␊
[18:43:59:243] malloc size 36␍␊
[18:43:59:243] allocate memory at 0x20012954, size: 48␍␊
[18:43:59:249] hit␍␊
[18:43:59:249] malloc size 11, but align to 12␍␊
[18:43:59:253] allocate memory at 0x20012984, size: 24␍␊
[18:43:59:256] hit␍␊
[18:43:59:256] malloc size 7, but align to 8␍␊
[18:43:59:259] allocate memory at 0x2001299c, size: 24␍␊
[18:43:59:262] hit␍␊
[18:43:59:267] release memory 0x20012984, size: 24␍␊
[18:43:59:270] malloc size 76␍␊
[18:43:59:270] allocate memory at 0x200129b4, size: 88␍␊
[18:43:59:274] hit␍␊
[18:43:59:274] malloc size 10, but align to 12␍␊
[18:43:59:279] allocate memory at 0x20012984, size: 24␍␊
[18:43:59:282] hit␍␊
[18:43:59:282] malloc size 8␍␊
[18:43:59:285] allocate memory at 0x20012a0c, size: 24␍␊
[18:43:59:288] hit␍␊
[18:43:59:288] malloc size 10, but align to 12␍␊
[18:43:59:293] allocate memory at 0x20012a24, size: 24␍␊
[18:43:59:293] hit␍␊
[18:43:59:293] release memory 0x20012984, size: 24␍␊
[18:43:59:299] malloc size 11, but align to 12␍␊
[18:43:59:302] allocate memory at 0x20012984, size: 24␍␊
[18:43:59:305] hit␍␊
[18:43:59:305] malloc size 11, but align to 12␍␊
[18:43:59:308] allocate memory at 0x20012a3c, size: 24␍␊
[18:43:59:311] hit␍␊
[18:43:59:315] release memory 0x20012984, size: 24␍␊
[18:43:59:318] malloc size 56␍␊
[18:43:59:318] allocate memory at 0x20012a54, size: 68␍␊
[18:43:59:324] hit␍␊
[18:43:59:324] malloc size 168␍␊
[18:43:59:324] allocate memory at 0x20012a98, size: 180␍␊
[18:43:59:331] hit␍␊
[18:44:00:900] big mem!, 1024, 21340, 21340, 75804␍␊
[18:44:00:903] malloc size 1024␍␊
[18:44:00:903] allocate memory at 0x20012b4c, size: 1036␍␊
[18:44:00:911] hit␍␊
[18:44:00:911] malloc size 72␍␊
[18:44:00:911] allocate memory at 0x20012f58, size: 84␍␊
[18:44:00:917] hit␍␊
[18:44:02:277] big mem!, 1320, 22460, 22460, 75804␍␊
[18:44:02:280] malloc size 1320␍␊
[18:44:02:280] allocate memory at 0x20012fac, size: 1332␍␊
[18:44:02:289] hit␍␊
[18:44:02:289] malloc size 72␍␊
[18:44:02:289] allocate memory at 0x200134e0, size: 84␍␊
[18:44:02:295] hit␍␊
[18:44:02:295] malloc size 640␍␊
[18:44:02:295] allocate memory at 0x20013534, size: 652␍␊
[18:44:02:301] hit␍␊
[18:44:02:301] malloc size 68␍␊
[18:44:02:301] allocate memory at 0x200137c0, size: 80␍␊
[18:44:02:305] hit␍␊
[18:44:02:305] malloc size 64␍␊
[18:44:02:305] allocate memory at 0x20013810, size: 76␍␊
[18:44:02:310] hit␍␊
[18:44:02:314] malloc size 640␍␊
[18:44:02:314] allocate memory at 0x2001385c, size: 652␍␊
[18:44:02:319] hit␍␊
[18:44:02:319] malloc size 52␍␊
[18:44:02:319] allocate memory at 0x20013ae8, size: 64␍␊
[18:44:02:324] hit␍␊
[18:44:02:330] malloc size 168␍␊
[18:44:02:330] allocate memory at 0x20013b28, size: 180␍␊
[18:44:02:334] hit␍␊
[18:44:02:337] malloc size 512␍␊
[18:44:02:337] allocate memory at 0x20013bdc, size: 524␍␊
[18:44:02:341] hit␍␊
[18:44:02:344] malloc size 168␍␊
[18:44:02:344] allocate memory at 0x20013de8, size: 180␍␊
[18:44:02:347] hit␍␊
[18:44:02:351] malloc size 128␍␊
[18:44:02:351] allocate memory at 0x20013e9c, size: 140␍␊
[18:44:02:354] hit␍␊
[18:44:02:358] malloc size 68␍␊
[18:44:02:358] allocate memory at 0x20013f28, size: 80␍␊
[18:44:02:361] hit␍␊
[18:44:02:361] malloc size 32␍␊
[18:44:02:364] allocate memory at 0x20013f78, size: 44␍␊
[18:44:02:369] hit␍␊
[18:44:02:369] malloc size 168␍␊
[18:44:02:369] allocate memory at 0x20013fa4, size: 180␍␊
[18:44:02:375] hit␍␊
[18:44:02:375] malloc size 256␍␊
[18:44:02:375] allocate memory at 0x20014058, size: 268␍␊
[18:44:02:382] hit␍␊
[18:44:02:382] malloc size 168␍␊
[18:44:02:382] allocate memory at 0x20014164, size: 180␍␊
[18:44:02:386] hit␍␊
[18:44:02:386] malloc size 128␍␊
[18:44:02:386] allocate memory at 0x20014218, size: 140␍␊
[18:44:02:390] hit␍␊
[18:44:02:390] malloc size 544␍␊
[18:44:02:393] allocate memory at 0x200142a4, size: 556␍␊
[18:44:02:397] hit␍␊
[18:44:02:397] malloc size 168␍␊
[18:44:02:401] allocate memory at 0x200144d0, size: 180␍␊
[18:44:02:404] hit␍␊
[18:44:04:565] big mem!, 4096, 28052, 28052, 75804␍␊
[18:44:07:573] malloc size 4096␍␊
[18:45:34:931] allocate memory at 0x20014584, size: 4108␍␊
```
值得注意的是到最后分配的也只有32K左右，实际的大小为75K：
`[18:44:04:565] big mem!, 4096, 28052, 28052, 75804␍␊`
50%的内存使用率真的可以吗。

另外的一个问题是，我是用了rt_malloc_sethook，但是不是每次分配内存都会调用这个hook，以上调试记录中打印hit的就是调用了hook的，有很多分配并没有调用这个hook。。。

BernardXiong, 2020-01-01:
1. slab内存管理算法适合于超过1MB内存情况，甚至是推荐>4MB才使用；
2. mem算法，每分配一块内存，开销是12字节；（32位处理器情况）
3. memheap算法，每分配一块内存，开销是24字节；

具体效率可以看你分配数据块的情况。另外，如果是mem算法，可以打开memtrace来查看具体分配情况，当然打开memtrace后，分配一块内存的开销会从12字节增长到16字节。如果是64位处理器的情况，开销则会从24字节增长到32字节；

mttbx, 2020-01-02:
12字节的开销应该不是问题，大多数内存分配算法大概也是这么大，假设分配100次，也就1K左右的开销。我换了个内存管理算法，现在程序就可以跑了。建议还是再看看这个算法的实现，可能会有一些最坏分配效率的问题，比如采用指数增长来进行块分配，可能会导致超过一倍的资源被浪费。我没有了解过mem和slab算法，不过显然效率是有一定问题的。

mttbx, 2020-01-02:
我有时间试试memtrace，之前没用过。

mttbx, 2020-01-02:
调试以后发现是堆栈溢出了，没问题了

closed 2020-01-02 by mttbx