RT-Thread/rt-thread #3305: 个人觉得需要讨论一下的一个地方

shao7936626, 2020-01-01:
情况是这样的：我用at组件的时候，明明能连上网，但是一连上网日志打印AT SOCKET(0) receive timeout，然后模块明明正常，但是却不能联网，tcp也连不上了；一直调整驱动程序，发现驱动没有问题，单独用at指令也能发送数据。最后发现at_device驱动里面，有check_link_status，每次check后，会调用net_dev组件中low_level_set_link_status，然后**问题就找到了**，这边调用了sal组件中的sal_check_netdev_internet_up；然后居然这边check internet的方式是向**link.rt-thread.org:8101**，发送RT_VERSION这些。这边我觉得不太严谨，虽然开销不是很大，但是每次都往上面发数据，流量是一个问题，第二个是如果link.rt-thread.org:8101关掉了，会不会影响到设备的稳定性？我个人试下来，他连不上link.rt-thread.org:8101的话，会把网卡设置为没有联网状态，导致上层应用无法正常使用

BernardXiong, 2020-01-01:
可能这个地方是需要调整下

@armink @Lawlieta

Lawlieta, 2020-01-02:
你好，关于上述网卡状态的问题，首先需要明确一点 **LINK_UP/DOWN 状态会影响设备网络通讯，而 INTERNET_UP/DOWN 状态目前只用于显示设备是否成功连接外网，并不会影响网络数据通讯。**

基于这点，我回答一下你上述提到的几个问题和疑惑：

1.  low_level_set_link_status 会调用 sal_check_netdev_internet_up 每次都要发送数据检查网卡 internet_up 网卡连接状态？
A: 这里可能你理解有点错误，并不是每次 low_level_set_link_status 函数调用时都会调用 sal_check_netdev_internet_up 检查 internet_up 状态，而是只有 LINK 状态从 LINK_DOWN 切换到 LINK_UP 状态的时候才会去检查一次 INTERNET_UP 状态，正常设备也不会频繁的改变 LINK 状态，所以这里你说的流量的开销并不是问题。

2.  如果link.rt-thread.org:8101关掉了，会不会影响到设备的稳定性？
如上面所述，INTERNET_UP/DOWN 状态只用于显示设备外网连接状态，并不会影响设备网络通讯，如果 link.rt-thread.org:8101 服务器关掉或者出问题，也之后影响 ifconfig 命令时设备状态显示 INTERNET_DOWN，设备网络通讯还是会正常进行。不会存在您所述“导致上层应用无法正常使用”。

closed 2020-01-02 by shao7936626