RT-Thread/rt-thread #3315: imxrt1052-fire-pro menuconfig BSP_USING_ETH

scJingP, 2020-01-03:
你好，我当前使用的开发板是野火 RT 1052; 我需要移植网卡模块。基于"网络协议栈驱动移植笔记" 文档，进行移植。
有以下问题请教：
1. 针对 imxrt1052-fire-pro 这个板子，为什么 menuconfig 中没有"Enable Ethernet"这个选项。
2.我有拿当前板子与stm32进行对比，在驱动实现 rt_imxrt_eth_init 这个函数。stm32 menuconfig 中支持"Enable Ethernet"这个选项，而 imxrt1052-fire-pro 不支持。
3.如果我的分析有误，还请说明我该如何正确移植 imxrt1052-fire-pro 的网卡模块。最终实现网络通讯。
期待您的回复，谢谢！

DavidLin1577, 2020-01-04:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    Component没有Network选项？我看了rt-thread4,.0.1的bsp包。1050有相关配置项，比如Network interface device，lwip版本选择，如果是新推出的开发板，即使没有上传到GitHub，野火光盘理论上也有相关demo，请知晓。
    
        
            
        
    
    
     
    
        
                    
            
                
                    
                            
                                
                            
                            
                                linpeng1577
                            
                    
                        
                            
                                    linpeng1577@gmail.com
                            
                        
                
            
        
        
    
    
        签名由
        网易邮箱大师
        定制

scJingP, 2020-01-08:
问题已解决

scJingP, 2020-01-08:
野火RT1052网卡功能移植已完成, 但有问题还请官方注意, 在实际运行中会卡死在 drv_eth.c 文件中 _enet_config 函数下 "ENET_Deinit(imxrt_eth_device.enet_base);" 代码处. 注释后程序正常运行, ping正常.
@DavidLin1577

closed 2020-01-08 by scJingP