RT-Thread/rt-thread #3330: rt-thread+lwip ENET_RXD0信号到 recvfrom 时延到ms 还能是实时？

scJingP, 2020-01-14:
开发环境：
RT-Thread: v4.0.2(master)
SOC: i.MX RT1050
Board: 野火 RT1052

系统环境：
lwip(2.0.2) + 测试程序（接收(recvfrom)到数据后马上输出IO信号）

测试方法：
探头一：ENET_RXD0引脚
探头二：IO信号引脚
测试这两个信号的时延

目的：网卡接收到数据后立即进行业务处理，这个过程中的时延在us级，抖动控制在100us内

现象：目前经过测试发现时延最高到9ms, 最低时延为97us

问题：如果使“最高时延-最低时延<=100us”

DavidLin1577, 2020-01-14:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    引用资料解释实时操作系统：    实时操作系统指系统的计算正确性不仅取决于计算的逻辑正确性，还取决于产生结果的时间，如果未满足系统的时间约束，则认为系统失效。    言归正传：系统的时间约束不是指执行时间的快慢，而是有方法在预设时间完成任务。
    
        你的问题：如何使“最高时延-最低时延<=100us”，测得时延最高到9ms，这个与你定的网络线程优先级，以及外部中断有关。假如你的开发板已经有多个线程在执行，而网络接收线程的优先级又比其他线程低，此时高优先级线程触发会抢占，导致低优先级线程无法在你期望的时间运行。解决手段：1.提高网络接收线程的优先级2.使用中断形式接收网络数据。以上个人经验，供参考。
    
     
    
        
                    
            
                
                    
                            
                                
                            
                            
                                linpeng1577
                            
                    
                        
                            
                                    linpeng1577@gmail.com
                            
                        
                
            
        
        
    
    
        签名由
        网易邮箱大师
        定制

BernardXiong, 2020-01-15:
这类问题请到论坛上进行讨论，or please write it in English.

closed 2020-01-15 by BernardXiong