RT-Thread/rt-thread #3360: GCC_VERSION macro redefined when building with sqlite3

djndl1, 2020-02-05:
The auto-generated `cconfig.h` has a `GCC_VERSION` macro that is defined by `tools/gcc.py`, which is conflicting with the `GCC_VERSION` defined in `sqlite3.c`. After removing this macro from `cconfig.h` manually, it compiled successfully. The generated `GCC_VERSION` doesn't seem to be used in any part of rt-thread, and considering the string that it expands to, I don't see how it can used in a source file. It cannot be used in a preprocessor expressions, as the compiling error message indicates. It can be fixed pretty easily.

```bash
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/packages/sqlite-v3.19.3/sqlite3.o
In file included from /home/djn/CodeSpace/rt-thread/include/libc/libc_signal.h:19:0,
                 from /home/djn/CodeSpace/rt-thread/include/rtdef.h:467,
                 from /home/djn/CodeSpace/rt-thread/include/rtthread.h:24,
                 from packages/sqlite-v3.19.3/sqlite3.c:20728:
./cconfig.h:15:0: warning: "GCC_VERSION" redefined
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
 
packages/sqlite-v3.19.3/sqlite3.c:227:0: note: this is the location of the previous definition
 # define GCC_VERSION (__GNUC__*1000000+__GNUC_MINOR__*1000+__GNUC_PATCHLEVEL__)
 
In file included from /home/djn/CodeSpace/rt-thread/include/libc/libc_signal.h:19:0,
                 from /home/djn/CodeSpace/rt-thread/include/rtdef.h:467,
                 from /home/djn/CodeSpace/rt-thread/include/rtthread.h:24,
                 from packages/sqlite-v3.19.3/sqlite3.c:20728:
packages/sqlite-v3.19.3/sqlite3.c: In function 'sqlite3Get4byte':
./cconfig.h:15:21: error: token ""6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"" is not valid in preprocessor expressions
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
                     ^
./cconfig.h:15:21: note: in definition of macro 'GCC_VERSION'
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
packages/sqlite-v3.19.3/sqlite3.c: In function 'sqlite3Put4byte':
./cconfig.h:15:21: error: token ""6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"" is not valid in preprocessor expressions
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
                     ^
./cconfig.h:15:21: note: in definition of macro 'GCC_VERSION'
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
packages/sqlite-v3.19.3/sqlite3.c: In function 'sqlite3AddInt64':
./cconfig.h:15:21: error: token ""6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"" is not valid in preprocessor expressions
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
                     ^
./cconfig.h:15:21: note: in definition of macro 'GCC_VERSION'
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
packages/sqlite-v3.19.3/sqlite3.c: In function 'sqlite3SubInt64':
./cconfig.h:15:21: error: token ""6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"" is not valid in preprocessor expressions
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
                     ^
./cconfig.h:15:21: note: in definition of macro 'GCC_VERSION'
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
packages/sqlite-v3.19.3/sqlite3.c: In function 'sqlite3MulInt64':
./cconfig.h:15:21: error: token ""6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"" is not valid in preprocessor expressions
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
                     ^
./cconfig.h:15:21: note: in definition of macro 'GCC_VERSION'
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from packages/sqlite-v3.19.3/sqlite3.c:29720:0:
packages/sqlite-v3.19.3/rtthread_vfs.c: In function '_rtthread_vfs_open':
packages/sqlite-v3.19.3/rtthread_vfs.c:271:9: warning: unused variable 'eType' [-Wunused-variable]
     int eType = flags & 0xFFFFFF00;  /* Type of file to open */
         ^~~~~
In file included from /home/djn/CodeSpace/rt-thread/include/libc/libc_signal.h:19:0,
                 from /home/djn/CodeSpace/rt-thread/include/rtdef.h:467,
                 from /home/djn/CodeSpace/rt-thread/include/rtthread.h:24,
                 from packages/sqlite-v3.19.3/sqlite3.c:20728:
packages/sqlite-v3.19.3/sqlite3.c: At top level:
./cconfig.h:15:21: error: token ""6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"" is not valid in preprocessor expressions
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
                     ^
./cconfig.h:15:21: note: in definition of macro 'GCC_VERSION'
 #define GCC_VERSION "6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
packages/sqlite-v3.19.3/sqlite3.c:20252:20: warning: 'sqlite3OsShmMap' defined but not used [-Wunused-function]
 SQLITE_PRIVATE int sqlite3OsShmMap(
                    ^~~~~~~~~~~~~~~
packages/sqlite-v3.19.3/sqlite3.c:20249:20: warning: 'sqlite3OsShmUnmap' defined but not used [-Wunused-function]
 SQLITE_PRIVATE int sqlite3OsShmUnmap(sqlite3_file *id, int deleteFlag){
                    ^~~~~~~~~~~~~~~~~
packages/sqlite-v3.19.3/sqlite3.c:20246:21: warning: 'sqlite3OsShmBarrier' defined but not used [-Wunused-function]
 SQLITE_PRIVATE void sqlite3OsShmBarrier(sqlite3_file *id){
                     ^~~~~~~~~~~~~~~~~~~
packages/sqlite-v3.19.3/sqlite3.c:20243:20: warning: 'sqlite3OsShmLock' defined but not used [-Wunused-function]
 SQLITE_PRIVATE int sqlite3OsShmLock(sqlite3_file *id, int offset, int n, int flags){
                    ^~~~~~~~~~~~~~~~
scons: *** [build/packages/sqlite-v3.19.3/sqlite3.o] Error 1
scons: building terminated because of errors.

```

BernardXiong, 2020-02-06:
Thank you for your feed back. The `GCC_VERSION` in cconfig.h is not used at present.

We can change the name in the next version.

BernardXiong, 2020-02-07:
Hello, I submit a PR, #3364. Please check it. Thanks.

closed 2020-02-07 by djndl1