RT-Thread/rt-thread #3388: Does rt-thread have any API for timed lock of mutex?

djndl1, 2020-02-18:
Currently I am trying to implement C++11 threading support based on rt-thread's POSIX interface. I've finished `<thread>` part and most of `<mutex>` but I just found out it's got no timed mutex support. Normally on POSIX-conformant systems, `try_lock_for` and `try_lock_until` are implemented by wrapping around `pthread_mutex_timedlock` but it's not available on rt-thread and I've found no other functions for this. Therefore, `timed_lock` and `recursive_timed_lock` are currently not implemented. Is there any plan to implement this function?

BernardXiong, 2020-02-19:
👍 You can PR a draft firstly, thank you.

djndl1, 2020-03-02:
Currently, some of the code is borrowed from libstdc++ and libc++ and there are some bugs already found during unit test. I might have to rewrite some to avoid legal issues. Also, I'm not sure if implementing C++ threads using pthread is really suitable for rt-thread as I'm not familiar with real-time OS. PR won't be made within the near two months.

mysterywolf, 2022-01-21:
https://github.com/RT-Thread/rt-thread/tree/master/components/cplusplus/cpp11

closed 2022-01-21 by mysterywolf